FreeSpacenav/spacenavd #3: Errors: “already using device”

fbruetting, 2018-09-05:
I tried to get my SpaceMouse Wireless (256f:c62e an 256f:c62f) to work on Fedora 28. Sadly, I just got no reaction and the first output below. When both the USB cable and the wireless receiver were plugged in, I got the second output below. With 0.6-7 it was about the same.

```
sudo spacenavd -v -d
Spacenav daemon 9d3d7da
failed to open config file /etc/spnavrc: No such file or directory. using defaults.
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62f]: "3Dconnexion SpaceMouse Wireless Receiver" (/dev/input/event22) 
adding device.
device name: 3Dconnexion SpaceMouse Wireless Receiver
  Number of axes: 6 (6a 0r)
  Number of buttons: 2
  Axis 0 value range: -350 - 350 (fuzz: 0)
  Axis 1 value range: -350 - 350 (fuzz: 0)
  Axis 2 value range: -350 - 350 (fuzz: 0)
  Axis 3 value range: -350 - 350 (fuzz: 0)
  Axis 4 value range: -350 - 350 (fuzz: 0)
  Axis 5 value range: -350 - 350 (fuzz: 0)
using device: 3Dconnexion SpaceMouse Wireless Receiver (/dev/input/event22)
trying to open X11 display ":0"
   XAUTHORITY=/run/user/1000/gdm/Xauthority
unhandled event: 17

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62f]: "3Dconnexion SpaceMouse Wireless Receiver" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless Receiver (/dev/input/event22)

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62f]: "3Dconnexion SpaceMouse Wireless Receiver" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless Receiver (/dev/input/event22)

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62f]: "3Dconnexion SpaceMouse Wireless Receiver" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless Receiver (/dev/input/event22)

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62f]: "3Dconnexion SpaceMouse Wireless Receiver" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless Receiver (/dev/input/event22)
adding dev event for device: /dev/input/event22
```

```
sudo spacenavd -v -d
Spacenav daemon 9d3d7da
failed to open config file /etc/spnavrc: No such file or directory. using defaults.
Device detection, parsing /proc/bus/input/devices
trying alternative detection, querying /dev/input/ devices...
  trying "/dev/input/." ...   trying "/dev/input/.." ...   trying "/dev/input/event21" ...   trying "/dev/input/by-id" ...   trying "/dev/input/event20" ...   trying "/dev/input/event19" ...   trying "/dev/input/event18" ...   trying "/dev/input/event17" ...   trying "/dev/input/event16" ...   trying "/dev/input/event15" ...   trying "/dev/input/event14" ...   trying "/dev/input/event13" ...   trying "/dev/input/event12" ...   trying "/dev/input/by-path" ...   trying "/dev/input/event11" ...   trying "/dev/input/event10" ...   trying "/dev/input/event9" ...   trying "/dev/input/event8" ...   trying "/dev/input/mouse1" ...   trying "/dev/input/event7" ...   trying "/dev/input/event6" ...   trying "/dev/input/event5" ...   trying "/dev/input/mouse0" ...   trying "/dev/input/event4" ...   trying "/dev/input/event3" ...   trying "/dev/input/event2" ...   trying "/dev/input/event1" ...   trying "/dev/input/event0" ...   trying "/dev/input/mice" ... failed to find any supported devices
trying to open X11 display ":0"
   XAUTHORITY=/run/user/1000/gdm/Xauthority

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
trying alternative detection, querying /dev/input/ devices...
  trying "/dev/input/." ...   trying "/dev/input/.." ...   trying "/dev/input/event21" ...   trying "/dev/input/by-id" ...   trying "/dev/input/event20" ...   trying "/dev/input/event19" ...   trying "/dev/input/event18" ...   trying "/dev/input/event17" ...   trying "/dev/input/event16" ...   trying "/dev/input/event15" ...   trying "/dev/input/event14" ...   trying "/dev/input/event13" ...   trying "/dev/input/event12" ...   trying "/dev/input/by-path" ...   trying "/dev/input/event11" ...   trying "/dev/input/event10" ...   trying "/dev/input/event9" ...   trying "/dev/input/event8" ...   trying "/dev/input/mouse1" ...   trying "/dev/input/event7" ...   trying "/dev/input/event6" ...   trying "/dev/input/event5" ...   trying "/dev/input/mouse0" ...   trying "/dev/input/event4" ...   trying "/dev/input/event3" ...   trying "/dev/input/event2" ...   trying "/dev/input/event1" ...   trying "/dev/input/event0" ...   trying "/dev/input/mice" ... failed to find any supported devices

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62e]: "3Dconnexion SpaceMouse Wireless" (/dev/input/event22) 
adding device.
device name: 3Dconnexion SpaceMouse Wireless
  Number of axes: 6 (6a 0r)
  Number of buttons: 2
  Axis 0 value range: -350 - 350 (fuzz: 0)
  Axis 1 value range: -350 - 350 (fuzz: 0)
  Axis 2 value range: -350 - 350 (fuzz: 0)
[... 133 lines cut ...]
handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62e]: "3Dconnexion SpaceMouse Wireless" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event22)

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62e]: "3Dconnexion SpaceMouse Wireless" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event22)

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62e]: "3Dconnexion SpaceMouse Wireless" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event22)

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62e]: "3Dconnexion SpaceMouse Wireless" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event22)

handle_hotplug called
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c62e]: "3Dconnexion SpaceMouse Wireless" (/dev/input/event22) 
already using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event22)
```

jtsiomb, 2018-09-05:
Please try to explain the problem you're reporting clearly. Are you saying that it doesn't work at all, or that it works but produces a lot of log output? 

The logfile clearly shows the device being detected, so if your report is about spacenavd not sending any events to applications, please describe what you tried, and ideally also try using the example programs which come with libspnav to make sure.

If your report is about the hotplug log output, is it that the hotplug events are in quick succession, resulting in high CPU usage, or are they something like one every few minutes and you're just reporting about redundant logging?

fbruetting, 2018-09-06:
It sadly doesn’t work at all, and I have the feeling that the mouse restarts every 5 seconds (would also explain the hotplug events). I never dis/reconnected the device, and this issue doesn’t occur in wireless mode ⇒ so this is made up. I did nothing else than starting spacenavd and connecting the mouse.

The third log was produced in just a few seconds, so the hotplug rate is some very few (1…3 seconds).

closed 2019-01-28 by jtsiomb

alextac98, 2020-04-27:
I have also come across this. The device itself is not faulty since it works perfectly fine in Windows. It seems like the device is disconnecting and reconnecting repeatedly. Is there any other advice you may have?

jtsiomb, 2020-04-27:
Unfortunately I don't have any device which exhibits this problem, so I can't be sure what the cause may be. As I said, I think it might be a problem with the kernel driver. Check dmesg, and if there are lots of disconnects and reconnects in the kernel log, I suggest finding the maintainers of the HID driver (I think it's mainly Greg Kroah Hartman, but look for a development mailing list or something) and ask them.

alextac98, 2020-05-06:
For the sake of documentation:
I am using the SpaceMouse Wireless. When attempting to use a direct connection with USB, it doesn't work. It does work, however in wireless mode with the included dongle. If I find some time, I will try to pinpoint the issue.