FreeSpacenav/spacenavd #7: 100% cpu usage when not able to accept on unix socket

flokli, 2019-01-19:
I'm currently working on getting spacenavd to work and properly packaged into NixOS.

I got it to build, and quickly executed it as non-root user. It forks away, then burns 100% cpu in the background.

Running it with `-d` shows the following error:
> error while accepting connection on the UNIX socket: Socket operation on non-socket

So I guess it has problems setting up the socket. And yes, `/var/run/` is not writable by the user. Can spacenavd check if it was able to set up the socket, and just exit with non-zero exit code if it fails, instead of trying again and again?

jtsiomb, 2019-01-20:
> Running it with `-d` shows the following error:
> 
> > error while accepting connection on the UNIX socket: Socket operation on non-socket

That's strange. If it's a matter of permissions in the socket directory, the bind call on line 60 (proto_unix.c) should fail, leaving lsock (global socket fd) to its default -1 value. Then in spnav.c line 134 the socket would not be added to the select list if it's invalid. Also handle_uevents, before the accept call does check for an invalid socket and bails. So the only way to end up going as far as the accept call which produces the error message you mentioned, is if we managed to create the socket without any errors, and then for some reason that file descriptor stopped corresponding to a socket, which is strange to say the least.

> So I guess it has problems setting up the socket. And yes, `/var/run/` is not writable by the user.

Spacenavd is not designed to be run by an unpriviledged user. It's a daemon, and needs to run as root during system startup from init.

If you wish to perform some further tests, to see if we can figure something out, add a few debug prints to see what's the file descriptor number we're creating at init, and what's passed to handle_uevents later on. So at the end of init_unix before the return statement add: `fprintf(stderr, "DBG: init_unix socket: %d\n", s);` and at the start of send_uevent right before the lsock == -1 check, add the same line with send_uevent instead of init_unix. Then find both DBG lines in the log and post them here.

jtsiomb, 2019-01-20:
The error is not checked on purpose, because we can still function through the X11 protocol handler without having a usable UNIX domain socket, and vice versa. See my previous answer on how this should still not produce the busy loop you're describing.

flokli, 2019-01-20:
Sorry for not fully reading through your previous comment!

I added the debug statements, get a `DBG: init_unix socket: 4`, a lot of `DBG: handle_uevents socket: 0` (and nothing from `send_uevent`):

```
$ strace -vvv -e write,bind,socket -fff result/bin/spacenavd 2>&1
[…]
write(2, "failed to find any supported dev"..., 37) = 37
socket(AF_NETLINK, SOCK_DGRAM, NETLINK_KOBJECT_UEVENT) = 3
bind(3, {sa_family=AF_NETLINK, nl_pid=20981, nl_groups=0xffffffff}, 12) = 0
socket(AF_UNIX, SOCK_STREAM, 0)         = 4
bind(4, {sa_family=AF_UNIX, sun_path="/var/run/spnav.sock"}, 110) = -1 EACCES (Permission denied)
write(2, "failed to bind unix socket: /var"..., 67) = 67
write(2, "DBG: init_unix socket: 4\n", 25) = 25
socket(AF_UNIX, SOCK_STREAM|SOCK_CLOEXEC, 0) = 5
write(2, "DBG: handle_uevents socket: 0\n", 30) = 30
write(2, "error while accepting connection"..., 84) = 84
write(2, "DBG: handle_uevents socket: 0\n", 30) = 30
write(2, "error while accepting connection"..., 84) = 84
```

jtsiomb, 2019-01-21:
Ok this is literally impossible with the current source code of spacenavd. If bind returns -1, it returns imediately, and therefore cannot possibly reach the "DBG: init_unix socket" print. I should also have closed the socket when bind fails, as it stands we're leaking a file descriptor, but that's a minor issue, and not related to the problem you're experiencing.

Please make sure you are using the latest version of spacenavd from the git repository.

flokli, 2019-01-21:
Right, I was testing with the latest release, 0.6 /  a757f62.

Should have mentioned that ;-)

I'll check with latest master, also now containing the fd leak mentioned above.

flokli, 2019-01-21:
master (4da551cfe21c394993120d04c5373d0140d05792) does not show this behaviour.

Can you craft a new release containing the fixes?

jtsiomb, 2019-01-21:
I think I will, pretty soon. There's one minor thing I wanted to add before release. Until then, just use the latest git code.

closed 2019-01-21 by jtsiomb

Sohalt, 2019-12-18:
Any update on when the release will happen?

jtsiomb, 2019-12-18:
I don't even remember what I wanted to add and if I added it or not... so fuck it, let's make a release. Doing it now.