FreeSpacenav/spacenavd #11: failed to find any supported devices

codejamninja, 2019-02-07:
I am getting the following error when I try and use this driver with my 3DConnexion mouse on Ubuntu 18.10.

```
failed to find any supported devices
```

I have gone through the following tutorial and gotten the mouse to work with the xcube demo, so I know the mouse is connected correctly.

https://wiki.archlinux.org/index.php/3D_Mouse

jtsiomb, 2019-02-07:
I need much more information than that. For starters post the spacenavd log file `/var/log/spnavd.log`. If it's huge with repeated identical messages, make sure to post from the start of the log to the first couple of the repeating errors.

codejamninja, 2019-02-07:
I was using the `spacenavd` from apt-get and it didn't work, but it would spit out logs. When I compiled this code, it didn't spit out logs, so I thought it was broken. It turns out it works though, so the logging must have been changed since the precompiled version was published.

closed 2019-02-07 by codejamninja