FreeSpacenav/spacenavd #15: Spacenavd grabs cadmouse as well as spacemouse if using 1 receiver

gsocker, 2019-12-03:
I bought a  wireless spacemouse/cadmouse kit. With this setup, both are paired to the same universal receiver (USB ID 0x256f, 0xc652). When starting spacenavd, it will grab the mouse from the X server since both show up under the same 3DConnexion receiver. 

I did a quick fix by modifying the "match_usbdev" code to ignore devices with "Mouse" in the description. A better way to do this would probably be to filter on the handler line containing "mouse" instead.

jtsiomb, 2019-12-04:
Are you testing this with the current git version of spacenavd? Because recently we've added a blacklist, which includes the USB ID you mentioned (`devid_blacklist` at `dev.c` line 229). The whole thing was discussed in pull request #4

gsocker, 2019-12-06:
I had to modify the blacklist first to remove the USB device id. After doing that, and modifying the code not to match the cadmouse, everything works. The X server still has the cadmouse, and Freecad/Blender pick up the spacemouse.
I have 3 entries in /proc/bus/input//devices for the receiver; there appears to be one entry each for the receiver, the spacemouse, and the cadmouse.
I've attached the contents of /proc/bus/input/devices for the receiver, as well as the lsusb output
[lsusb.txt](https://github.com/FreeSpacenav/spacenavd/files/3934395/lsusb.txt)
[receiver_devices.txt](https://github.com/FreeSpacenav/spacenavd/files/3934385/receiver_devices.txt)

jtsiomb, 2019-12-15:
I wouldn't like to hack this blindly without being able to test anything. Care to share your changes in the form of a patch or a pull request, so I can review exactly what you did?

gsocker, 2019-12-20:
[universal_receiver.txt](https://github.com/FreeSpacenav/spacenavd/files/3990715/universal_receiver.txt)

Patch that got it working for me is attached.

jtsiomb, 2019-12-21:
Thank you I appreciate it. I've also received a pull request for the same issue which seems very similar. I'll review both changes and see what must be done.

jtsiomb, 2019-12-21:
I just merged the pull request. It was pretty much this, but the string matching was slightly more specific and I thought less likely to cause unintended problems down the line. Could you please check the current git HEAD and let me know if it works correctly for you?

gsocker, 2019-12-24:
Yes,the current head works for me as well.

closed 2019-12-24 by gsocker

jtsiomb, 2019-12-25:
Excellent. Thank you.