FreeSpacenav/spacenavd #18: [master] Running configure from release tarbals shows Git errors

hartwork, 2020-01-31:
Hi!

The current configure script does not cope well with being run from a release archive when no Git folder exists:
https://github.com/FreeSpacenav/spacenavd/blob/8acf883d97ea30a098621ac7128e5d40ba90f496/configure#L43-L49

```console
# tar xf spacenavd-0.7.tar.gz
# cd spacenavd-0.7
# ./configure 
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
configuring spacenavd - 
  prefix: /usr/local
  optimize for speed: yes
  include debugging symbols: yes
  x11 communication method: yes
  use hotplug: yes

creating Makefile ...
creating config.h

Done. You can now type make (or gmake) to compile spacenavd.
```

Is that known and intended?

jtsiomb, 2020-01-31:
No, it's a bug.

jtsiomb, 2020-02-01:
I added a script to prepare releases in the future, that modifies the VER line in configure, to not rely on git. I'll use it to make a bugfix release (0.7.1) soon which shouldn't have this problem. Leaving this issue open until then.

jtsiomb, 2020-02-08:
Release 0.7.1 fixed this issue.

closed 2020-02-08 by jtsiomb