FreeSpacenav/spacenavd #19: [master] ./configure --disable-x11 causes link error src/dev.c:91: undefined reference to `drop_xinput'

hartwork, 2020-02-01:
Full log (with GCC 8.2.0):

```console
# cd "$(mktemp -d)"

# git clone https://github.com/FreeSpacenav/spacenavd.git
[..]

# cd spacenavd/

# ./configure --disable-x11
configuring spacenavd - v0.7-7-gd16ef85
  prefix: /usr/local
  optimize for speed: yes
  include debugging symbols: yes
  x11 communication method: no
  use hotplug: yes

WARNING: you have disabled the X11 interface, the resulting daemon won't be compatible with applications written for the proprietary 3Dconnexion daemon (3dxserv)!

creating Makefile ...
creating config.h

Done. You can now type make (or gmake) to compile spacenavd.

# make
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/cfgfile.c -o src/cfgfile.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/client.c -o src/client.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/dev.c -o src/dev.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/dev_serial.c -o src/dev_serial.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/dev_usb.c -o src/dev_usb.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/dev_usb_darwin.c -o src/dev_usb_darwin.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/dev_usb_linux.c -o src/dev_usb_linux.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/dummy_usb.c -o src/dummy_usb.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/event.c -o src/event.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/hotplug_darwin.c -o src/hotplug_darwin.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/hotplug_linux.c -o src/hotplug_linux.o
src/hotplug_linux.c: In function ‘poll_timeout’:
src/hotplug_linux.c:139:4: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
    write(poll_pipe, &sig, 1);
    ^~~~~~~~~~~~~~~~~~~~~~~~~
src/hotplug_linux.c: In function ‘handle_hotplug’:
src/hotplug_linux.c:94:2: warning: ignoring return value of ‘read’, declared with attribute warn_unused_result [-Wunused-result]
  read(hotplug_fd, buf, sizeof buf);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/kbemu.c -o src/kbemu.o
src/kbemu.c:62: warning: ISO C forbids an empty translation unit [-Wpedantic]
 #endif /* USE_X11 */
 
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/logger.c -o src/logger.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/magellan/smag.c -o src/magellan/smag.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/magellan/smag_comm.c -o src/magellan/smag_comm.o
src/magellan/smag_comm.c: In function ‘smag_write’:
src/magellan/smag_comm.c:119:3: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
   write(fd, buf + i, 1);
   ^~~~~~~~~~~~~~~~~~~~~
src/magellan/smag_comm.c:122:2: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
  write(fd, "\r", 1);
  ^~~~~~~~~~~~~~~~~~
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/magellan/smag_detect.c -o src/magellan/smag_detect.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/magellan/smag_event.c -o src/magellan/smag_event.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/proto_unix.c -o src/proto_unix.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/proto_x11.c -o src/proto_x11.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/serial/sball.c -o src/serial/sball.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/serial/sballserial.c -o src/serial/sballserial.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/spnavd.c -o src/spnavd.o
src/spnavd.c: In function ‘daemonize’:
src/spnavd.c:252:2: warning: ignoring return value of ‘chdir’, declared with attribute warn_unused_result [-Wunused-result]
  chdir("/");
  ^~~~~~~~~~
src/spnavd.c:268:2: warning: ignoring return value of ‘dup’, declared with attribute warn_unused_result [-Wunused-result]
  dup(1);
  ^~~~~~
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/spnavd_win32.c -o src/spnavd_win32.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/xdetect_freebsd.c -o src/xdetect_freebsd.o
gcc -pedantic -Wall -g -O2 -fno-strict-aliasing -fcommon -I./src -I/usr/local/include  -c src/xdetect_linux.c -o src/xdetect_linux.o
src/xdetect_linux.c:167: warning: ISO C forbids an empty translation unit [-Wpedantic]
 #endif /* __linux__ */
 
gcc -o spacenavd src/cfgfile.o src/client.o src/dev.o src/dev_serial.o src/dev_usb.o src/dev_usb_darwin.o src/dev_usb_linux.o src/dummy_usb.o src/event.o src/hotplug_darwin.o src/hotplug_linux.o src/kbemu.o src/logger.o src/magellan/smag.o src/magellan/smag_comm.o src/magellan/smag_detect.o src/magellan/smag_event.o src/proto_unix.o src/proto_x11.o src/serial/sball.o src/serial/sballserial.o src/spnavd.o src/spnavd_win32.o src/xdetect_freebsd.o src/xdetect_linux.o -L/usr/local/lib  
/usr/lib/gcc/x86_64-pc-linux-gnu/8.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: src/dev.o: in function `init_devices':
/tmp/tmp.Ra4QnfmuQ6/spacenavd/src/dev.c:91: undefined reference to `drop_xinput'
collect2: error: ld returned 1 exit status
make: *** [Makefile:20: spacenavd] Error 1
```

jtsiomb, 2020-02-01:
The non-X11 build is rarely if ever tested. Thanks for reporting this. I think commit 3168b1d fixes this, so I'm closing the issue. Feel free to re-open if the problem persist (I was too lazy to actually test it to be honest).

closed 2020-02-01 by jtsiomb

hartwork, 2020-02-01:
Looks good — thanks for the fix!