FreeSpacenav/spacenavd #27: no log file entries with spnavrc settings enabled

dlarue, 2020-11-14:
On Ubuntu 18.04.3 recently built spacenavd v0.7.1 I have set in my /etc/spnavrc file:
logfile = /var/log/spnavd.log
log = file, syslog

but I've not seen any creation of the /var/log/spnav.log file and no contents after I've created the file and restarted the daemon.

dlarue, 2020-11-14:
I had script for systemctl script was using spacenavd -d -v and it seems in daemon mode it will not log anything. I figured this out when I realized the instructions for installing included running setup_init and that's where I saw it was setting up a systemd init file with only "spacenav -v".

But while it now will log to /var/log/spnavd.log  that log shows errors in the spnavrc file for setting up the log file. It does not recognize the logfile and log options... weird.
Log on startup looksl ike this:

Spacenav daemon v0.7.1-8-gd330773
reading config file: /etc/spnavrc
unrecognized config option: logfile
unrecognized config option: log
Device detection, parsing /proc/bus/input/devices
found usb device [46d:c626]: "3Dconnexion SpaceNavigator" (/dev/input/event17) 
adding device.
device name: 3Dconnexion SpaceNavigator
  Number of axes: 6 (6a 0r)
  Number of buttons: 2
  Axis 0 value range: -500 - 500 (fuzz: 0)
  Axis 1 value range: -500 - 500 (fuzz: 0)
  Axis 2 value range: -500 - 500 (fuzz: 0)
  Axis 3 value range: -500 - 500 (fuzz: 0)
  Axis 4 value range: -500 - 500 (fuzz: 0)
  Axis 5 value range: -500 - 500 (fuzz: 0)
using device: 3Dconnexion SpaceNavigator (/dev/input/event17)

jtsiomb, 2020-11-15:
The logging method selection was changed from a configuration option to a commandline argument, and I don't think you can specify both a file and syslog, I think it's one or the other.

The `-d` option means "do *not* daemonize". By default spacenavd runs as a daemon.

Leaving this bug report open, to remember and remove references to log/logfile configuration options from the example spnavrc file.

closed 2020-11-15 by jtsiomb

dlarue, 2020-11-15:
IIRC I tried spacenavd --help and got nothing to show what options were available. my man page on spacenavd was of no help either. I guess I was too lazy to look for main.c to see what argc/argv were passed and tested.

I felt like I was chasing my tail where I had spacenavd running for years via SystemCtl but when trying to get logging working nothing was logging or doing any output( -d -v options ).  It wasn't until I grabbed the git repo and built then realized the instructions said to also run setup_init so I did that after creating a dummy /etc/inittab file. THEN I got logging but now when I tried to run systemctl it would fail to start the daemon with just -v which is what they systemd init script ran with.

I love using the Space Navigator and REALLY appreciate the work done to keep it working on Linux since 3Dconnexion have been far less than Linux friendly/helpful.

jtsiomb, 2020-11-16:
Yes, the option for help is actually `-h`. It would be more user-friendly to accept the long form too, and/or output usage information when it encounters invalid options.

As for not being able to start it with systemctl, I'm afraid I can't help. I have absolutely no familiarity or interest in systemd. I am including a contributed systemd init file under `contrib/`, but I'm not in a position to maintain or support it in any meaningful way.