FreeSpacenav/spacenavd #28: SpaceNavigator buttton 0 not mapping Escape key

dlarue, 2020-11-14:
On Ubuntu 18.04.3  running spacenavd v0.7.1 I have enabled kbmap0 = Escape in /etc/spnavrc but do not see the effect in either FreeCAD nor Blender.

I see the buttons mapped and operate when I disable that setting in /etc/spnavrc in both applications and I see that button stop operating the mapping once I enable 'kbmap0 = Escape' in /etc/spnavrc but it does not initiate an Escape key operation.

Tested by using a left-mouse-button-click to open a menu then clicking the button0 on the space navigator to clear the menu with the Escape key.

In FreeCAD I even tried removing the Button0 mapping just to see if that changed anything and it did not.

dlarue, 2020-11-14:
After realizing my logging issues( another issue ) were because my systemctl script was starting the daemon with the -d option I fixed that and now the Escape and Control_L button remapping is both being logged but the Control_L seems to not function in FreeCAD but does seem to work in Blender.

Spacenav daemon v0.7.1-8-gd330773
reading config file: /etc/spnavrc
unrecognized config option: logfile
unrecognized config option: log
Device detection, parsing /proc/bus/input/devices
found usb device [46d:c626]: "3Dconnexion SpaceNavigator" (/dev/input/event17) 
adding device.
device name: 3Dconnexion SpaceNavigator
  Number of axes: 6 (6a 0r)
  Number of buttons: 2
  Axis 0 value range: -500 - 500 (fuzz: 0)
  Axis 1 value range: -500 - 500 (fuzz: 0)
  Axis 2 value range: -500 - 500 (fuzz: 0)
  Axis 3 value range: -500 - 500 (fuzz: 0)
  Axis 4 value range: -500 - 500 (fuzz: 0)
  Axis 5 value range: -500 - 500 (fuzz: 0)
using device: 3Dconnexion SpaceNavigator (/dev/input/event17)
trying to open X11 display ":0"
   XAUTHORITY=/tmp/xauth-1000-_0
unhandled event: 17
unhandled event: 17
unhandled event: 4
mapping ``Escape'' to keysym 65307
unhandled event: 4
unhandled event: 4
mapping ``Control_L'' to keysym 65507
unhandled event: 4

jtsiomb, 2020-11-15:
I'm not sure modifier keys are really going to work with the way spaceanvd attempts to send key events (XSendEvent). And also some programs will ignore such synthetic X events even for keys that will work (like Escape). A more reliable way to achieve this would be to use the Xtest extension if available.

dlarue, 2020-11-15:
Esc and Ctrl both work on Blender and the Esc key modifier works on FreeCAD. 

I started this whole escapade when someone was trying to use user.conf configurations in FreeCAD to get one of his spaceball pro buttons to emit the Ctrl modifier.  I couldn't get that to work.

jtsiomb, 2020-11-15:
I just tested and can verify that blender works with all the key mappings I tried, while FreeCAD doesn't seem to respond to ctrl and shift, but accepts escape.

Since this works on some programs and not others, it all comes down to how each program is handling events. Specifically the issue with FreeCAD probably has to do with how it handles modifier keys, since it's not a case of rejecting synthetic events altogether.

Re-implementing this keymapping functionality with Xtest might or might not help in this case, I'm not sure. But as this is not really a bug, I'm marking it as a feature request to try out Xtest at some point and see if it helps.

jtsiomb, 2022-01-20:
XTest support was merged recently (git commit: 4c503c2d1d733228e9b728c2b6c8ef44689b5bd3) and is now the default method for sending key events, if spacenavd is built with Xtest support (see logfile for "Using XTEST to send key events" message to verify). Can someone please try it and report if this issue has been resolved or not?

dlarue, 2022-01-20:
I'll try to get to it tomorrow.

jtsiomb, 2022-03-31:
I just tested it, and it seems to work fine now.

closed 2022-03-31 by jtsiomb

lninio, 2024-09-07:
Hi! It seems I has found the error in "configure" in string 197. Please add quotes in _echo_ instruction.
[image]

jtsiomb, 2024-09-07:
> Hi! I think I found an error in "configure" on line 197. Please add quotation marks to the _echo_ instruction.

What does this have to do with mapping the Escape key?