FreeSpacenav/spacenavd #29: [0.8] Release tarball lacks hardcoded version 0.8, ends up with #define VERSION ""

hartwork, 2020-12-01:
After running configure with the 0.8 release tarball, I end up with `#define VERSION ""` in `./src/config.h` rather than  `#define VERSION "0.8"`. So it seems broken, is that known?

jtsiomb, 2020-12-01:
I need to come up with a scheme that works outside of the git repo too. Maybe if git returns an error code, it should try to use the part of the current directory after `spacenavd-` as a version string.

I don't think this is important enough to warrant a bugfix 8.1 release. The version is only used in the logfiles. But I'll try to remember to fix this before the next release.

hartwork, 2020-12-01:
> I need to come up with a scheme that works outside of the git repo too. Maybe if git returns an error code, it should try to use the part of the current directory after `spacenavd-` as a version string.

The rest of the world hardcodes versions in the code at one place, ideally only one. Why does that need to be avoided at high cost?

> The version is only used in the logfiles.

Doesn't seem true to me:
```
# spacenavd -version  # with my Gentoo patch
spacenavd 0.8
```

jtsiomb, 2020-12-01:
Because it's almost guaranteed that I'll forget to change the hardcoded version number when it's time for the next release.

hartwork, 2020-12-01:
Why guaranteed? That approach seems to work well 99% of the time for many other projects. The more releases you do, the more of a habit bumping versions before releases becomes. No?

jtsiomb, 2020-12-01:
No.

hartwork, 2020-12-01:
Why no?

jtsiomb, 2020-12-01:
fixed in revision d3a3aa0

closed 2020-12-01 by jtsiomb