FreeSpacenav/spacenavd #32: Recommended test if it's working?

MarcelRobitaille, 2021-01-23:
I followed [this guide](https://mattjones.tech/2019/03/26/using-3dconnexion-spacemouse-with-linux/) that says to check that it's working I can open blender and click the shoulder buttons. I clicked all the buttons on the mouse in blender and nothing happens.

When I run `spacenavd -v -d`, I get no output, `/var/log/spnavd.log` doesn't exist, but the blue light on my mouse lights up.

Is there a recommended way to test if it's working? I know the FAQ talks about xinput, but how do I actually test that?

Sorry for the stupid question, I'm new to this space mouse business.

MarcelRobitaille, 2021-01-23:
Oh yeah, I don't have anything similar to FAQ 9 in my `xorg.conf`.

MarcelRobitaille, 2021-01-23:
Ok, I read some other issues and I see that `libspnav` comes with examples. I compiled `cube`, but when I run it, I get `failed to connect to the space navigator daemon` even though it is running (no output no logs as I said before).

jtsiomb, 2021-01-23:
The log file doesn't exist, because you're running with `-d`, which means "do not daemonize", so all messages are printed in the console rather than the log file. It's the same messages that would go to the log file.

If the AF_UNIX method of connecting to spacenavd works, spacenavd works. The problem with programs using the X11 interface is that most likely spacenavd can't connect to your X server. See FAQ #10: http://spacenav.sourceforge.net/faq.html#faq10

Closing this as it's not really a bug. But feel free to continue asking followup questions if you can't make it work.

closed 2021-01-23 by jtsiomb

MarcelRobitaille, 2021-01-23:
Thanks for the response.

At first (before reading `spacenavd -h`) I ran without `-d` and still no log file. Maybe because no messages were logged in either case.

I read that FAQ, but I thought it didn't apply to me. I have just been testing, not starting on boot, so I use `sudo`. I will try the fix all the same.

It may not be a bug, but maybe the error messages and docs could be improved for new users? I can make a pull request to the docs once I get it working if you would be open to that?

MarcelRobitaille, 2021-01-23:
Ok, I had `/root/.Xauthority` as an empty file. I overwrote it with `sudo ln -sf ~/.Xauthority /root/.Xauthority` and now cube example, blender, and freecad are working. Thanks for your help!

jtsiomb, 2021-01-23:
Sure, any suggestions for improving the documentation are welcome. I don't know if pull requests are the best way to do that though, given that there's no documentation in the source tree.

MarcelRobitaille, 2021-01-23:
Oh I see.

Maybe FAQ 10 could be updated? It didn't seem to reflect my situation.

Also, any idea why I didn't get `failed to open X11 display` logged?

jtsiomb, 2021-01-23:
I'm sure you did but failed to notice.

MarcelRobitaille, 2021-01-23:
I don't think so. Like I said, I wasn't getting any output at all. It's not like it could have gotten lost within thousands of lines.

MarcelRobitaille, 2021-01-24:
On my other machine, I get no logs, but even `simple_af_unix` gives `failed to connect to the space navigator daemon`. Should I open a new issue?

jtsiomb, 2021-01-24:
I don't understand your no logs thing. Are you saying that the log file is empty? Or that it's not created at all? Make sure you run spacenavd as root. It's not meant to be run by the user as needed, it's meant to run from init as root.

For easier debugging, if you just want to run it and see what happens, pass the -d argument, and whatever would end up in the log file, will be printed on the terminal.

MarcelRobitaille, 2021-01-24:
The log file wasn't created at all. I then tried with `-d`, and absolutely nothing was printed. Maybe that's because `start_syslog` is [only called in `daemonize`](https://github.com/FreeSpacenav/spacenavd/blob/d3a3aa017a281f8875eb1f14018290795c0dd0d5/src/spnavd.c#L294), so any calls to `logmsg` do nothing because `use_syslog` is false.

I know it's meant to be run as root on boot, but I am just trying to get it working and changing flags to see the result. I am running as root.

jtsiomb, 2021-01-24:
syslog codepaths shouldn't affect anything unless you start with `-l syslog`.

There seems to be a regression when it comes to logging on the terminal when running with `-d` because that use case doesn't get much testing. I'll look into it. But I can't reproduce your issue of not logging to the log file. You may try to pass a different path for the logfile with `-l <logfile path>`, and see if that helps.

MarcelRobitaille, 2021-01-25:
Yeah, I don't know what was going on with the log file either. Ever since the second time I ran it, I've been using `-d`. Now when I run without `-d`, I get my log file. I'm sorry, I know that's not helpful. I checked today though and I am 100% sure that file did not exist and that I had run it without `-d`. Maybe the only time I ran it without `-d`, I forgot `sudo` and it couldn't write to `/var/`.

Anyway, on my other system, the examples aren't connecting. I run `sudo ./spacenavd -v`. I can check that it's running with `ps`. I am `tail -w`'ing the log file, which shows:
```
Spacenav daemon v0.8-1-gd3a3aa0
reading config file: /etc/spnavrc
adding device.
```
I try `./simple_af_unix` and get
```
connect failed: Permission denied
failed to connect to the space navigator daemon
```
Since it's a permission issue, I tried with `sudo`, but same result.

I'm sorry my question thread turned into a bug report. I can open a new one if you want.