FreeSpacenav/spacenavd #45: Buttons on SpaceMouse Enterprise do nothing

dyaso, 2021-07-20:
The [device has 33 buttons](https://3dconnexion.com/uk/product/spacemouse-enterprise/#start), none of which currently work. 

I'd very much like it if the Shift and Control keys could be made to work, but have no idea how to help -- I've tried running `xev` but no events seem to be sent.

jtsiomb, 2021-07-20:
libspnav comes with an example program (`libspnav/examples/simple`) which prints all the events received by spacenavd. You can use that to test the buttons, and to Use that to test the buttons, and to figure out which button corresponds to which number, so that you can set up key mappings in `/etc/spnavrc`. See the example configuration file (`spacenavd/doc/example-spnavrc`) the section starting with "Map buttons to keyboard keys".

dyaso, 2021-07-20:
Thanks for your help!

When I make a remapping in `/etc/spnavrc` it doesn't do anything, but the button stops being reported as pressed by the `examples/simple_af_unix` program.

jtsiomb, 2021-07-20:
Yes of course that's significant. If spacenavd can't connect to the X server, then it can't send keyboard events. Also the UNIX socket protocol will work (`simple_af_unix`) but the X11-based protocol won't (`simple_x11`). See FAQ number 10: http://spacenav.sourceforge.net/faq.html#faq10

dyaso, 2021-07-21:
Apologies for not having read the FAQ, and wasting your time. Button remapping works now, barring several buttons which all identify as #0.

How plausible would it be to have buttons send Ctrl+Z / Shift+Ctrl+Z key combinations with a single press? I realize I'd probably have to modify `spacenavd` somewhere.

closed 2021-07-21 by dyaso

jtsiomb, 2021-07-21:
It's not possible currently to send multiple key events, but there's no reason why it couldn't be done. If you wish open a feature request (marked enhancement) about it.