FreeSpacenav/spacenavd #46: Example "simple" gets stuck in "spnav_wait_event(&sev)"

yottasyno, 2021-08-11:
Hello,

I connected the 3Dconnexion mouse via usb, started the spacenavd and launched the example "simple". Unfortunately it is not printing to the console when I move the 3D mouse.

Can someone help with this?

Thanks a lot!

jtsiomb, 2021-08-11:
Which 3dconnexion mouse? Are you running `simple_x11` or `simple_af_unix`? Does `/var/log/spnavd.log` have anything interesting in it?

yottasyno, 2021-08-12:
I restarted everything. Plugged the mouse in and out. Now I managed to connect and print the values to the console. Not sure what the problem was. I tried different daemons and libraries and maybe created some conflicts. I'm giving all information anyways, maybe there is some useful feedback:

Thanks a lot for the quick response! Thanks for this alternative to the official SDK!

**Mouse:** 3Dconnexion SpaceMouse Wireless
**Program:** simple_af_unix while spacenavd is running OR simple_x11 while 3DxWare 1.8.0 daemon is running
**/var/log/spnavd.log:**
```
Spacenav daemon v0.8-13-gc528816
reading config file: /etc/spnavrc
failed to open config file /etc/spnavrc: No such file or directory. using defaults.
failed to find any supported devices
Using XSendEvent to send key events

OR

Spacenav daemon v0.8-13-gc528816
reading config file: /etc/spnavrc
failed to open config file /etc/spnavrc: No such file or directory. using defaults.
adding device.
device name: 3Dconnexion SpaceMouse Wireless
failed to grab the device: Device or resource busy
using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event25)
Using XSendEvent to send key events
failed to find any supported devices
read error: No such device
removing device: 3Dconnexion SpaceMouse Wireless
failed to turn LED off
failed to find any supported devices
failed to find any supported devices
failed to find any supported devices
...
...
failed to find any supported devices
failed to find any supported devices
failed to find any supported devices
adding device.
device name: 3Dconnexion SpaceMouse Wireless
failed to grab the device: Device or resource busy
using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event19)
failed to find any supported devices
read error: No such device
removing device: 3Dconnexion SpaceMouse Wireless
failed to turn LED off
failed to find any supported devices
failed to find any supported devices
failed to find any supported devices
...
...
```

closed 2021-08-12 by yottasyno