FreeSpacenav/spacenavd #47: [Feature] Change Sensitivity live

hasechris, 2021-09-22:
Hi,

I'm an owner of an SpacePilot, which has some buttons around the spacemouse. I got pretty much all working in FreeCAD, so first thank you very much. This made my transition from Autodesk Inventor (they cancelled my subscription after 10 years in the education program) to FreeCAD so much nicer.

I just miss one feature here. I have two buttons to increase or decrease the sensitivity. Would this be possible in the future?

Best rergards
hasechris

jtsiomb, 2021-09-22:
Yes, that's quite useful, and I thought about it before. I'll make sure to implement it sooner rather than later.

jtsiomb, 2021-09-23:
Added a mechanism (commit 9f748d1), to bind actions to buttons through the configuration file. Buttons with bound actions are not reported to clients. Currently implemented actions: sensitivity up/down/reset. See example-spnavrc for syntax.

Closing the feature request, but any feedback, and ideas for other bindable actions, appreciated.

closed 2021-09-23 by jtsiomb