FreeSpacenav/spacenavd #48: Support for Houdini?

slvr2314, 2021-11-05:
I used spacenavd with Blender and it works perfectly fine, but I couldn't get it to work with Houdini. I don't know if I'm missing something or if spacenavd just doesn't support Houdini right now. But if it's the latter, could spacenavd support Side Effects Houdini in the future?

jtsiomb, 2021-11-05:
Does it work with the proprietary 3dconnexion driver?

slvr2314, 2021-11-05:
Ah! Sorry, I know it works on Windows. But I haven't yet tried the Linux driver.

jtsiomb, 2021-11-05:
There are a few distinct possibilities. First of all, since blender works, and houdini doesn't, we can safely assume houdini, unlike blender, doesn't use the spacenavd protocol.

  - If houdini supports 6dof input on GNU/Linux using the 3dconnexion API, it should work with spacenavd automatically, unless there's a regression in spacenavd. Please try to run the proprietary driver to see if that's the case.

  - If houdini supports 6dof input on GNU/Linux by directly accessing the device, it should work (probably only for USB devices) without spacenavd. Try stopping spacenavd and try to run houdini again.

  - If houdini doesn't support 6dof input on GNU/Linux at all, there's nothing we can do. You'll have to ask them to add support.

slvr2314, 2021-11-06:
I'm having trouble installing the proprietary driver because it's asking for libXm.so.3, which is a part of openmotif 2.2.x, but the version on my repo (Manjaro) is openmotif 2.3.x, which comes with libXm.so.4, and I can't find it anywhere else either. Any advice?

jtsiomb, 2021-11-06:
Google pointed me to this: https://motif.ics.com/open-motif/download/openmotif-223targz

The lazy approach would be to try and symlink libXm.so.4 as libXm.so.3 and see if it works or crashes.

slvr2314, 2021-11-06:
Unfortunately, the proprietary driver doesn't support Houdini. :( I'm guessing it's because it was last updated back in 2014. At the time, the Windows driver didn't support Houdini either. I can't believe they're still frequently updating the Windows and Mac drivers but have completely given up on Linux. Do "designers" not use Linux? :/
Thank you @jtsiomb and everyone else who contributed to this project. It might not be very useful to me rn, but it's still an incredible project! And who knows, maybe a future version of Houdini might use the spacenavd protocol 🤞.

jtsiomb, 2021-11-06:
It's not so much up to the driver, mine or 3dconnexion's, regardless of update frequency. You need to let the houdini developers know you want them to support 6dof devices on GNU/Linux. Point them to our project website in case the reason they don't do so is they percieve the 3dconnexion GNU/Linux driver as abandoned.

As this is probably Houdini's fault, I'm closing this bug report. Feel free to re-open if it turns out there's something we need to do to support it.

closed 2021-11-06 by jtsiomb