FreeSpacenav/spacenavd #56: Friendlier output when when have spacenav library, but aren't running the mouse daemon

luzpaz, 2022-02-04:
Is there a standard practice to make the output of spacenav lib more friendly when it comes to output

[image]

This a screenshot of FreeCAD invoked from the CLI. It's calling spacenav but not finding the daemon. 

Any suggestions?

t-paul, 2022-02-04:
I second that from OpenSCAD perspective. With the SpNav driver enabled but no deamon running the same error is shown on console and there seems to be no way to suppress that.

jtsiomb, 2022-02-04:
These messages have been removed from libspnav two years ago in commit 844338bd699969dfc9e4577d8d268e9902002cb6

closed 2022-02-04 by jtsiomb

jtsiomb, 2022-02-04:
I guess if you had to work around it in a previous version, you could do something hacky like:

    int tmpstderr = dup(2);
    close(2);
    open("/dev/null");
    ... call to spnav_open with silenced std errror output ...
    close(2);
    dup(tmpstderr);
    close(tmpstderr);

But that would be slightly absurd :)

t-paul, 2022-02-04:
Nice it's already solved. Linux distros usually only ship released versions, so it will only appear for users once a new release of the library happens.

jtsiomb, 2022-02-04:
I guess I could put out a minor release, I just didn't think this was an important enough change for it. I'll think about it.

luzpaz, 2022-02-05:
:+1: for minor release

luzpaz, 2022-02-05:
@jtsiomb would you kindly notate this ticket once a new release is made? I'd like to close the [downstream ticket](https://tracker.freecadweb.org/view.php?id=4711) to this and coordinate with the FreeCAD packagers

t-paul, 2022-02-05:
@luzpaz better (and automatic way):

Go to repository for the client library (this one is the server part).

Select `Watch` -> `Custom` -> `Releases` and you'll automatically receive a notification from github as soon as there's a release marked.

luzpaz, 2022-02-05:
TIL  
Thanks!

jtsiomb, 2022-02-06:
I just released libspnav 0.3 which contains this fix.