FreeSpacenav/spacenavd #58: Filter movements

albfan, 2022-02-08:
Hi, I just installed this userspace driver and it works out of the box.

I tried samples and cube is rotating as expected, but find hard to translate only or rotate only.

Then I tried old driver using https://wiki.archlinux.org/title/3D_Mouse and after some tricks:

link libXm.so.4 to libXm.so.3
run `./3dmouse root` with sudo

I was able to run that driver too.

[image]

If found it attaches two functions to buttons (filter translation and filter rotation, which are incompatible, only one can be active or none)

[image]

that helps to move only in right direction, but even they have a config to filter dominant axis, and that means that only clear action on rotation or translation is followed

[image]

Is it possible to add that to this driver? I tried with spnavcfg but that option is not available

[image]

I can work on that

[image]

just disabling sensitivity for translation or rotation we have same behaviour (and I see there's config for increase/decrease sensitivity (probably global) but I can adjust to have that feature)

filter is probably something more complex but probably just take first axis with more than X amount of change will do it

Do you think it is interesting to implement? I will do changes on config repo too (to setup buttons or activate this filters by default)

NOTE: kill window "magellan window" if for some reason config window for old driver is still active

jtsiomb, 2022-02-09:
I think a dominant axis mode will certainly be useful to have, and shouldn't be hard to implement. The only reason I never did that, was because I never liked that mode very much myself. But it would certainly be nice to have it as an option.

Masking translation or rotation completely is certainly trivial. I'm not sure it would be of much use if dominant axis mode was supported, but it doesn't hurt to have the option I guess.

I see that you provided a pull request, but haven't examined it yet. I'll respond about implementation specifics there if necessary.

albfan, 2022-02-10:
Now this is merged I can confirm "dominant axis filter" works by choosing from all the input the biggest value.

If you push right,back and up at same time, jumps from one to the other a little bit, but user most likely will not do that and movement is intuitive and smooth.

Working on it

albfan, 2023-01-28:
I'm revisiting the driver and all looks really smooth. config app looks great. Will refactor all changes and let you known

albfan, 2023-02-25:
Resolved!

closed 2023-02-25 by albfan