FreeSpacenav/spacenavd #69: Anyone able to get the spacenavd updated in Debian experimental/unstable package repo?

luzpaz, 2022-04-12:
Downstream FreeCAD project would really appreciate it. 
https://packages.debian.org/sid/source/spacenavd
https://salsa.debian.org/ropp-guest/spacenavd/activity

CC @RoPP

Lenglemetz, 2022-05-07:
I prefer a PPA or a snap. Maybe developers create this?

jtsiomb, 2022-05-07:
The spacenav project does not distribute binaries, only source code. And as for the debian packaging, there's not much I can do about it, so I'm closing this issue.

closed 2022-05-07 by jtsiomb

luzpaz, 2022-05-07:
Does anyone know Rodolphe Pelloux-Prayer ? Apparently they're the spacenavd debian salsa maintainer https://packages.debian.org/sid/source/spacenavd

I guess, I was hoping that someone who know this person could let them know. They haven't responded to email or the git issue ping.