FreeSpacenav/spacenavd #70: MacOS: failed to find any supported USB devices

chevdor, 2022-05-16:
I built from source, after setting `X11=no`. 

```
sudo spacenavd -v -d
Spacenav daemon v1.0-4-g2c5a443
reading config file: /etc/spnavrc
failed to find any supported USB devices
```

My SpaceMouse Compact is recognozed in the OS and works fine in Fusion 360 and Blender.
`spacenavd` however does not see to recognize it.

jtsiomb, 2022-05-16:
Unfortunately at this point only serial devices are supported on MacOS. I'll mark this as a feature request.

chevdor, 2022-05-17:
Ah ok, that explains the troubles then. Thanks for clearing that up.

luzpaz, 2022-06-07:
oh bummer

consi, 2023-10-17:
Hello,

Is there any possibility that this would be considered in future? I have 3dconnexion SpaceExplorer which support was dropped in macOS drivers but device is running perfectly fine

jtsiomb, 2023-10-18:
It's quite probable, but I can't give you a timeline. The likeliest scenario is that this will be automatically solved, when I get around to writing an alternative cross-platform version of the USB code based on libusb, which has been on my TODO list for years.

Of course 3rd party contributions are always welcome. If anyone wishes to work on this (either through libusb or through the platform-specific macos HID API) let me know, and I will help with the integration as much as I can.

kevinwright, 2023-12-15:
I'd be up for taking a look, not least because I'd like to see it working with the bluetooth spacemouse pro.

One query though... it looks like `dev_usb_darwin.c` has been in the source code for at least 2 years, and uses the platform-specific `IOKit/hid/IOHIDLib.h`. So I'm not entirely sure what's missing here

jtsiomb, 2023-12-15:
Sounds great!

`dev_usb_darwin.c` was my unfinished attempt at implementing USB support for macosx 11 years ago. I was using macos for a little while back then for work, and thought I'd give it a go, but it was never completed.

I don't remember what state I left it in, feel free to continue working on that code, or start from scratch, whichever you prefer.

luzpaz, 2024-10-17:
Any progress here ?

alextrical, 2025-04-26:
I'm going to be looking into this project in the next few weeks is ive aquired a "SpacePilot" and will try to help (my coding skils are somewhat lackluster for the languages required here.)

As for USB support, i wonder if it would be possible to use a project such as "[HID Remapper](https://github.com/jfedor2/hid-remapper)" to expose a USB devise as a USB Serial device?

jtsiomb, 2025-04-26:
That would be a horrible kludge, and not acceptable for merge into spacenavd. You can certainly try it for your own amusement though if you feel inclined.

alextrical, 2025-04-26:
Agreed on it being a kludge, It would definitely be best to get USB driver support working on Mac OS. I will see how far i get on a software only solution for USB.

alextrical, 2025-04-27:
So the what would be the best way to get USB support on MacOS in your opinion? Getting LibUSB support added over going for platform specific implementations?

So for a libUSB I'm guessing the follow needs to be done:
1. Add libusb Dependencies
2. Replace the current device detection code with libusb device enumeration
3. Replace the current device I/O operations with libusb calls
4. Create udev rules for Linux systems to provide appropriate access permissions

I think getting this working on Linux/Ubuntu first probably is my best bet, as I'm hoping that would allow me to get support/assitance while adding LibUSB support

alextrical, 2025-04-27:
Another alternative may be using the Python based input from here https://github.com/JakubAndrysek/PySpaceMouse but doesn't feel like the cleanest way of going about adding cross platform USB

jtsiomb, 2025-04-27:
libusb is probably a good fallback for unsupported platforms, but it should be a build-time option, not replace the platform-specific code outright. There might be advantages and disadvantages to each, which should be evaluated on each platform, to decide if there is a reason to have platform-specific code on that platform or not. Another option is the hidapi library, which is probably easier to get going, since it already handles all the USB HID-specifics.

udev rules are not necessary. spacenavd is meant to run as root and should have all the access it needs.

alextrical, 2025-05-05:
> Another option is the hidapi library, which is probably easier to get going, since it already handles all the USB HID-specifics.

OK, im happy to give it a go. 
I will setup a Ubuntu Linux build environment and try getting hidapi working on linux first, and then get it running on a Mac.
So far I've been able to detect a (emulated) SpaceMouse Pro on a M1 Mac, 

[image]

Im slowly working out what is needed for USB HID communication

CtrlC-Root, 2025-12-06:
I have an Apple laptop for work now and we use `libusb` for cross-platform USB access. I'd be happy to test or review any PRs. I don't have time for the next few months but I could take a stab at implementing this if no one else beats me to it. Commenting here so I can keep tabs on this.

smc8050, 2025-12-09:
For anyone reading this thread, I have found this project: [PySpaceMouse](https://github.com/JakubAndrysek/PySpaceMouse) which uses `hidapi` to connect to 3DConnexion spacemice, I tested it with a SpacePilot Pro and it works on a MacBookPro M3 with MacOS 26.1.
Maybe this would help someone in the future.