FreeSpacenav/spacenavd #71: Failed to open X11 display ":0.0"

tgmatt, 2022-07-31:
I know this issue has come up a lot and I've tried all the solutions currently out there but I'm still getting the X11 error:
```
trying to open X11 display ":0.0"
   XAUTHORITY=/root/.Xauthority
failed to open X11 display ":0.0"
```
The `/root/.Xauthority` is symlinked to my own one thusly:
```
lrwxrwxrwx 1 root root 22 Sep 18  2021 /root/.Xauthority -> /home/matt/.Xauthority
```
Running `sudo spnavd_ctl x11 start` throws the following output:
```
You must have an X server running before starting up spacenavd-X11 events.
```

My DISPLAY env var is set to `:1`, not sure why or what that means honestly. I tried changing it in spnavd_ctl to match, which "succeeds," but `spnavd.log` shows a similar error above with a completely different `XAUTHORITY`.

I'm running Pop_OS 22.04, everything up to date. spacenavd and libspnav have been compiled from source on version 1.0.

jtsiomb, 2022-07-31:
Interesting. The problem in your case is probably not the authority file, but rather that your X server is running as display 1, while spacenavd clearly tries to connect to display 0. Spacenavd will default to trying display 0 when the `DISPLAY` envirionment variable is missing when it's starting. So assuming that's the problem, try to start spacenavd with `DISPLAY=:1` in the environment.

Running `spnavd_ctl x11 start` is unnecessary. Spacenavd should be able to detect when the X server starts running and connect to it automatically.

tgmatt, 2022-07-31:
I modified the init file to include `DISPLAY=:1` but still seems to fail:
```
Spacenav daemon v1.0-4-g2c5a443
reading config file: /etc/spnavrc
failed to open config file /etc/spnavrc: No such file or directory. using defaults.
Device detection, parsing /proc/bus/input/devices
found usb device [256f:c652]: "3Dconnexion 3Dconnexion Universal Receiver" (/dev/input/event7) 
adding device (id: 0).
device name: 3Dconnexion 3Dconnexion Universal Receiver
  Number of axes: 6 (6a 0r)
  Number of buttons: 2
  Axis 0 value range: -350 - 350 (fuzz: 0)
  Axis 1 value range: -350 - 350 (fuzz: 0)
  Axis 2 value range: -350 - 350 (fuzz: 0)
  Axis 3 value range: -350 - 350 (fuzz: 0)
  Axis 4 value range: -350 - 350 (fuzz: 0)
  Axis 5 value range: -350 - 350 (fuzz: 0)
using device: 3Dconnexion 3Dconnexion Universal Receiver (/dev/input/event7)
  device flags: swap y-z invert y-z
trying to open X11 display ":1"
   XAUTHORITY=/root/.Xauthority
failed to open X11 display ":1"
waiting for the X socket file to appear
```
Very confused.

jtsiomb, 2022-07-31:
Does it work if you start spacenavd manually from a terminal after X has started? Use `-d` to stop if from daemonizing to see what's going on in the terminal itself.

tgmatt, 2022-07-31:
Running it as my own user I get:
```
failed to bind unix socket: /var/run/spnav.sock: Address already in use
trying to open X11 display ":1"
   XAUTHORITY=/run/user/1000/gdm/Xauthority
Using XSendEvent to send key events
```
and as root, I get:
```
trying to open X11 display ":1"
   XAUTHORITY=/run/user/1000/gdm/Xauthority
Using XSendEvent to send key events
```
If I change the `XAUTHORITY` to the one copied from my user, I get:
```
trying to open X11 display ":1"
   XAUTHORITY=/root/.Xauthority
Authorization required, but no authorization protocol specified
failed to open X11 display ":1"
waiting for the X socket file to appear
```
If I then run `xhost +` as root, then start it, I get:
```
trying to open X11 display ":1"
   XAUTHORITY=/root/.Xauthority
Using XSendEvent to send key events
```
but I don't get any output by pressing any buttons, and it doesn't work in FreeCAD either.

tgmatt, 2022-07-31:
OK so I think this is partially down to the 3dconnexion wireless adapter because if I connect it via USB directly, it records events in `spnavd.log`, but still doesn't work in FreeCAD. The wireless definitely used to work so I'm not really sure why that is no longer working but at this stage I'd settle for it just working via wired.

jtsiomb, 2022-07-31:
To test if it works or not, use the "simple" example program which comes with libspnav.  Specifically the simple example compiles into two executables: `simple_af_unix` and `simple_x11`. Both of them should print 6dof input events when you move or press/release buttons. If the first one doesn't work, probably nothing else will; either your device is dead or spacenavd is not running. If the second one doesn't work, it means spacenavd can't talk to the X server (xauth, wrong DISPLAY, etc). If both of them work, and the program you're trying to use can't use 6dof input, then the issue is with the program.

Of the multiple attempts you posted, at least two of them seem absolutely fine. You need to run as root, so the first one doesn't matter, it's clearly a permission issue for the socket. The second one seems fine, the third one failed to connect to the X server due to incorrect xauth, and the fourth one also seems fine.

Unrelated issue, but when you see "Using XSendEvent to send key events", you probably didn't have the Xtest library installed when building spacenavd, which means mapping buttons to keyboard events will not work reliably. If you care about that feature, compile with Xtest support.

tgmatt, 2022-08-02:
It seems to have been resolved by adding `DISPLAY=:1` to the init script and I'm having to use it wired as wireless isn't working for some reason. It's listed in `lsusb`, but whatever - I can work with it wired for now anway. Not sure why my display variable is `:1` and not `:0`, but alas I'm content with closing this issue now. Cheers 👍🏻

closed 2022-08-02 by tgmatt

jtsiomb, 2022-08-02:
Glad it works at least in wired mode. Ideally I would like to figure out why it doesn't work wireless too, but not having any of the wireless devices myself it's difficult to find out what's different. I will send an email to 3dconnexion to see if they're interested in donating a wireless device to the project.

jtsiomb, 2022-08-25:
Which device do you have? I asked 3Dconnexion to send me a wireless device try and debug the wireless issue, which has been reported before. They sent me a "SpaceMouse Wireless", which however worked fine both wired and wireless.

tgmatt, 2022-08-25:
I have the SpaceMouse Wireless, too. I've moved to using Onshape for my CAD because I just couldn't get on with FreeCAD, and I don't think the SpaceMouse works in the browser on Linux, does it?

jtsiomb, 2022-08-25:
Strange. I wonder why the wireless mode didn't work in your case.
Yes, onshape uses a new completely different API using websockets, which is not supported by spacenavd. There was some initial interest by a few people to work on implementing that, but so far it hasn't gone very far.