FreeSpacenav/spacenavd #73: Get Button Numbers

hasechris, 2022-10-30:
Hi,

could you build a mode which would write the pressed button numbers into the verbose log? Currently i have to guess the button numbers for the function mapping.

Greetings
hasechris

jtsiomb, 2022-10-31:
I don't think the log is the place for that. There are already multiple ways to determine button numbers:

  - Use the "simple" example application which comes with libspnav (under `examples/simple`), and prints out all events (axis and button) it receives from spacenavd while running.
  - Use the configuration GUI (spnavcfg), in the "Buttons" tab, which highlights and displays the button number for every button press (and also allows you to do all the mappings directly from the GUI).
  - Check the button documentation in the wiki: https://github.com/FreeSpacenav/spacenavd/wiki/Device-button-names Do not trust it blindly, but it's probably correct for the included devices. Please consider contributing button to number mapping info if you happen to have a device not listed there.

As it stands, I'll close this issue, assuming you didn't know those methods, but let me know if the above are not sufficient for what you need for some reason.

closed 2022-10-31 by jtsiomb

hasechris, 2022-10-31:
@jtsiomb 

Hrmpf... Cant explain how i didnt catch one of these methods/infos in the documentation. Thank you very much :-)