FreeSpacenav/spacenavd #85: Wireless Adapter Not Working

space-person, 2023-05-03:
So I compiled spacenavd 1.2, libspnav 1.1 and spnavcfg-1.0 from source, following the installation instructions, and also followed FAQ #10 from https://spacenav.sourceforge.net/faq.html and got SpaceNav to find my device, but no matter what I tried, I couldn't get input out of it.

Out of a whim, I decided to stop using the wireless dongle and switch to the hardwire USB and it magically started working.

Any ideas why that might be? Thanks in advance!

jtsiomb, 2023-05-03:
This has been reported before, but I was unable to reproduce it. I have a spacemouse wireless here, and it works with or without the cable. Can you post the USB vendor:device ids in both states (wired and wireless) ?

space-person, 2023-05-03:
I ran `lsusb` with the cable plug in, and then unplugged the cable and plugged in the wireless receiver and got these two outputs respectively.

Bus 001 Device 024: ID 256f:c62e 3Dconnexion SpaceMouse Wireless (cabled)
Bus 001 Device 025: ID 256f:c652 3Dconnexion Universal Receiver

Here is the output of `/var/log/spnav.log` after starting the daemon with the wireless receiver, then unplugging it and plugging in the wired cable.

> Spacenav daemon 1.2
reading config file: /etc/spnavrc
found usb device [256f:c652]: "3Dconnexion 3Dconnexion Universal Receiver" (/dev/input/event2) 
adding device (id: 0).
device name: 3Dconnexion 3Dconnexion Universal Receiver
  Number of axes: 6 (6a 0r)
  Number of buttons: 2 (evdev offset: 256)
  Axis 0 value range: -350 - 350 (fuzz: 0)
  Axis 1 value range: -350 - 350 (fuzz: 0)
  Axis 2 value range: -350 - 350 (fuzz: 0)
  Axis 3 value range: -350 - 350 (fuzz: 0)
  Axis 4 value range: -350 - 350 (fuzz: 0)
  Axis 5 value range: -350 - 350 (fuzz: 0)
failed to grab the device: Device or resource busy
using device: 3Dconnexion SpaceMouse Wireless (guess) (/dev/input/event2)
  device flags: swap y-z invert y-z
trying to open X11 display ":1"
   XAUTHORITY=/run/user/1000/gdm/Xauthority
Using XTEST to send key events
read error: No such device
removing device: 3Dconnexion SpaceMouse Wireless (guess) (id: 0 path: /dev/input/event2)
failed to turn LED off
found usb device [256f:c62e]: "3Dconnexion SpaceMouse Wireless" (/dev/input/event2) 
adding device (id: 1).
device name: 3Dconnexion SpaceMouse Wireless
  Number of axes: 6 (6a 0r)
  Number of buttons: 2 (evdev offset: 256)
  Axis 0 value range: -350 - 350 (fuzz: 0)
  Axis 1 value range: -350 - 350 (fuzz: 0)
  Axis 2 value range: -350 - 350 (fuzz: 0)
  Axis 3 value range: -350 - 350 (fuzz: 0)
  Axis 4 value range: -350 - 350 (fuzz: 0)
  Axis 5 value range: -350 - 350 (fuzz: 0)
using device: 3Dconnexion SpaceMouse Wireless (/dev/input/event2)
  device flags: swap y-z invert y-z

jtsiomb, 2023-05-03:
That's strange... it finds it, and starts querying its capabilities just fine, and then suddenly the device is gone. Does `dmesg` say anything useful when that happens?

space-person, 2023-05-04:
Right? I'm not sure why it's doing that. Even if I load up spnavcfg, it shows the correct model, correct number of axes, correct number of buttons, it just doesn't take input and I can't control the LED.

Here's the `dmesg` output after I plug the wireless dongle in.
`[83017.678000] usb 1-1: new full-speed USB device number 29 using xhci_hcd
[83018.010237] usb 1-1: New USB device found, idVendor=256f, idProduct=c652, bcdDevice= 1.06
[83018.010243] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[83018.010245] usb 1-1: Product: 3Dconnexion Universal Receiver
[83018.010247] usb 1-1: Manufacturer: 3Dconnexion
[83018.061563] input: 3Dconnexion 3Dconnexion Universal Receiver as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-1/1-1:1.0/0003:256F:C652.004A/input/input75
[83018.122212] hid-generic 0003:256F:C652.004A: input,hiddev0,hidraw0: USB HID v1.11 Multi-Axis Controller [3Dconnexion 3Dconnexion Universal Receiver] on usb-0000:03:00.0-1/input0
[83018.132470] hid-generic 0003:256F:C652.004B: hiddev1,hidraw1: USB HID v1.11 Device [3Dconnexion 3Dconnexion Universal Receiver] on usb-0000:03:00.0-1/input1
[83018.141298] hid-generic 0003:256F:C652.004C: hiddev2,hidraw2: USB HID v1.11 Device [3Dconnexion 3Dconnexion Universal Receiver] on usb-0000:03:00.0-1/input2
[83018.154303] hid-generic 0003:256F:C652.004D: hiddev3,hidraw3: USB HID v1.11 Device [3Dconnexion 3Dconnexion Universal Receiver] on usb-0000:03:00.0-1/input3
[83018.167318] hid-generic 0003:256F:C652.004E: hiddev4,hidraw4: USB HID v1.11 Device [3Dconnexion 3Dconnexion Universal Receiver] on usb-0000:03:00.0-1/input4
`

jtsiomb, 2023-05-04:
Yeah, that's exactly the same as the messages when I plug mine in. But mine doesn't disappear and keeps working.

You are running spacenavd as root, right? The error messages would be different if you weren't, so I assume that's not the issue... I don't know, I'll get back to you if I think of something.

space-person, 2023-05-04:
Wait... what? I'm getting it to work. I swapped it to a different USB port and it seems to work?! It's not a dead port though because my mouse works just fine on it.

I'm really happy it works, but I'm so curious and confused haha.

The only difference I can think of is that the port that didn't work is a USB 3.0 port and the one that did work was a USB 2.0 port. I don't know if that helps at all.

jtsiomb, 2023-05-05:
Interesting. Was it a hub or a built-in USB 3.0 port? I tried connecting mine to a built-in USB 3.0 port and it still works fine.  In any case this seems like a kernel issue since the device really disappears. I don't think there's anything I can do from the spacenavd side, so I'll go ahead and close the bug report.