FreeSpacenav/spacenavd #87: No movement if axis value doesn't change at all.

void-spark, 2023-07-17:
Ok, last issue was me not realizing something, this one seems legit :)

Manjaro, latest version, spacenavd and spnavcfg from AUR, libspnav from the repo:
[image]

I noticed it seemed at very gentle movements, close to the center of the device, movement seems very jerky and juddery.
This is noticeable both in Blender, and the examples from libspnav.

When keeping spnavcfg open, and handling the device very carefully, I noticed a clear pattern though.
As long as the value of a axis changes, movement continues.
But when the axis value doesn't change (but is not 0!!), so holding the device still off center,
movement stops completely!

So I guess somewhere there is something that only sends events when the device updates, but with the device still, no updates are sent.

Let me know if there is anything I can do to help investigate or solve this! :)

void-spark, 2023-07-17:
Oh for crying out loud.. :D
Ok, in spcnavrc there is a comment and a setting that repeats inputs when the axis is still :)
On my device this would make a great default, it doesn't drift from the 0 position, but I guess other models are different :)
Maybe something to put in the FAQ or front page ? :)

closed 2023-07-17 by void-spark

jtsiomb, 2023-07-17:
Yeah, some (most?) devices keep sending updates when the device is off-center, regardless of motion, others don't. So there's no default that would work on everything.

A FAQ entry would not be a bad idea, better yet would be a pre-set for the repeat interval based on device detection, but would need a relatively large sample of devices otherwise it's just guessing...

The repeat interval setting is also on the first page of the GUI as far as discoverability goes, but maybe it's not immediately obvious what it's supposed to do.

void-spark, 2023-07-17:
You're right! That just completely didn't stand out to me for some reason :)

[image]

void-spark, 2023-07-17:
Although, that said, does that actually work? Changing it from the GUI doesn't seem to persist for me..

void-spark, 2023-07-17:
I think in https://github.com/FreeSpacenav/spnavcfg/blob/master/src/ui.cc the setting of cfg.repeat might be missing :)

CtrlC-Root, 2023-07-17:
https://github.com/FreeSpacenav/spnavcfg/issues/32 seems related

jtsiomb, 2023-07-17:
You're absolutely right, I completely forgot about this.

jtsiomb, 2023-07-17:
Done. Implemented in the current git version of spnavcfg. The repeat interval was missing from the protocol handling across the board, so this latest spnavcfg also requires the latest git versions of both libspnav and spacenavd.