FreeSpacenav/spacenavd #88: [FEATURE REQUEST]: Execute command on spacemouse button press

ryryog25, 2023-08-21:
There might be a way to do this already, though I can't seem to find it, the internet has thus far disappointed me with regards to solutions, so anywho:

I've gotten very used to using Radial Menus with 3DxWare under macOS, and was trying to set up a similar system under linux, with my first thought being to assign the buttons to keypresses, then map those keypresses into something like Fly-Pie for its radial menus. However, due to the way Fly-Pie receives keystrokes, (as well as other programs tested) it doesn't recognize the simulated XTEST keypresses.

I've also tried QJoyPad and Input Mapper as workarounds, but they don't seem to work with the SpaceMouse's buttons.

After doing a fair bit of research, and trying to use GNOME-Pie instead, then realizing that it wouldn't work anyways because, since I'm running Wayland, keyboard shortcuts aren't a thing that play nicely, it came to my attention GNOME-Pie, as well as Fly-Pie support opening a menu via terminal command.

This would also be cleaner since I wouldn't have to dedicate specific keys for the radial menus, but as far as I can tell, spacenavd doesn't have a way to run a terminal command on the pressing of a spacemouse button.

If there's already a way to do this, I'd appreciate a pointer on that, and if not, I think this would be a very powerful feature in general. Not just for my use case, but being able to execute a command as I've described would open the door to much more powerful use of the spacemouse buttons, allowing the creation of custom scripts and such that could do virtually anything.

ryryog25, 2023-08-21:
It's also worth mentioning, although my programming knowledge and experience are relatively minimal, if a pointer as to how this feature could be implemented was given (I also haven't been able to find much documentation about the underlying spacenavd source, and I've not yet just opened up the source to look) I'd be willing to take a look if it's within my skillset.

jtsiomb, 2023-08-21:
Interesting. No, binding button presses to the execution of arbitrary commands is not currently supported.

To do this, at minimum you'd need to add a new config file option for the new kind of binding, and then fork/exec a shell with the specified command when the button is pressed.

There are a number of subtle issues that need to be considered however:
  - spacenavd is a daemon, which is supposed to run as root. The executed command probably shouldn't run as root. How would we specify the user the command should run as?
  - spacenavd being a daemon, is started by init before X has even started. Therefore DISPLAY is likely not set in spacenavd's environment, but should probably be set in the command's environment to be useful, and the command should be able to connect to the X server, so XAUTHORITY needs to be set as well.
  - For similar reasons, spacenavd's environment will have HOME, SHELL, etc set to the root values, not necessarilly what the user will expect.

None of these are insurmountable, but this whole mess points to the solution you're proposing being "not the Right Thing", at least for the purpose of spawning some kind of menu when you press a button. So let's go back to the original problem. What kind of menu do you want to spawn when you press a button, what sort of things would that menu contain? What would you do from that menu?

ryryog25, 2023-08-21:
Of course, find the best solution to the problem, not the best way to implement a proposed solution.

Under macOS (and Windows for that matter) 3DxWare supports radial menus (among other macros) that can be triggered when you press one of the side buttons. I use Fusion 360 for CAD, so in my case I had things like Extrude, Revolve, etc, in these menus, that way I rarely had to take my hands off the spacemouse or mouse, and could run all the functions with the radial menus.

It works by:
1. Pressing a button on the spacemouse configured to open a radial menu
2. A circle around the cursor appears, divided into (in my case) 4 sections, each corresponding to a given function.
3. The cursor can be moved in the direction of one of these sections and the function will run. (e.g. an extrude dialog/environment will open)
4. The menu disappears after the function has been triggered. 

Under linux, my plan was to use Fly-Pie (a generic radial menu tool) and create macros mapped to my keyboard shortcuts for each tool, similar to how I set it up under macOS. 

The goal is basically to turn my 2 buttons on the spacemouse into enough that I can access every commonly used tool without touching the keyboard or having to navigate the toolbar menus, making everything more efficient.

ryryog25, 2023-08-21:
While I agree this is a less than ideal solution, and obviously not complete, nor ubiquitous, I thought I should share it anyways, since it's a thing that exists now.

https://github.com/FreeSpacenav/spacenavd/commit/af8826e376d08040b89f9d67037acedebdd7f81a

In this commit I've added -ish support for what I way trying to do, added another action to the enumeration and run the given command with `runuser`

I'm running spacenavd under `systemctl --user` since that drastically simplified everything (the consequences of this I'm not sure... that's beyond me to be completely honest, as is anything I broke in the code, although it *seems* to work)

My `spacenavd.service` file then is located at `~/.local/share/systemd/user/spacenavd.service` and these are it's contents:
```
[Unit]
Description=3Dconnexion Input Devices Userspace Driver

[Service]
Type=forking
ExecStart=sudo spacenavd -v
StandardError=syslog

[Install]
WantedBy=default.target
```

My username is hardcoded into the runuser right this second, but besides that I think I've kept it relatively conformant.

In order to set up a command you just have to edit `/etc/spnavrc` and add `bnactN = run-command '[command]'`

The command runs correctly with the environment variables (so it seems, I've been testing it by launching `thunderbird` lol)

ryryog25, 2023-08-22:
Hmm... I see. Sounds like a project, but would be worth it if I could get it to work nicely.

Does sound like the cleaner way to do things though. I'll see if I can't set something up when I get a chance, appreciate the suggestions!

I'll close this issue for now, and open it back up if (when, more likely) I have questions or something of substance to share.

ryryog25, 2023-08-31:
I have finished a program for this purpose, which supports both keyboard macros and integration directly with Fusion360 running under WINE.

https://github.com/ryryog25/SpacenavdRadialMenus/

It has to be compiled from source and requires some other manual work to get it set up, along with some other inconveniences (needing to call `sudo killall ...`  to stop it is less than ideal) but it does seem to work pretty well, and if anyone else is interested in something like this (yes, all two of y'all) I'd be happy to help if there's any issues using it.

closed 2023-08-31 by ryryog25

ryryog25, 2023-08-31:
Suggestion noted, good point, I'll change that now. 

I think the offending bit of code is this (lines 99-103 in `main.cpp`)... black magic string of function calls that my setup (GNOME on Wayland) needed to make sure the window was always focused:

```
                menus[sev.button.bnum]->setWindowState(Qt::WindowMinimized);
                menus[sev.button.bnum]->activateWindow();
                menus[sev.button.bnum]->raise();
                menus[sev.button.bnum]->activateWindow();
                menus[sev.button.bnum]->show();
```

I needed this combination due to occasional weirdness since sometimes showing the window doesn't raise it to the top, and seems to be a common issue in wayland. If you'd be willing to try it for me, removing everything there but the call to `show()` (and maybe `activateWindow()`?) should fix that.

Besides that are you experiencing any other issues?

jtsiomb, 2023-08-31:
Yeap, with just activate and show it works correctly. I also tried leaving raise too, and it still works, and is probably a good idea.

I did not encounter any other issue, but I can't say I did any amount of extensive testing. The thing pops up, pressing the buttons sends keypresses, it works.

The only issues are probably things you already are aware of. Lack of configurability without editing the source code, no good way to terminate it, and also the attrocious polling loop checking for input every N milliseconds. But for a prototype all of these are fine.

For the termination thing, I'd probably try to add an icon to the desktop trey with a menu including "quit", if available, falling back to a standalone iconified window that pops the same menu when clicked if there is no tray.

ryryog25, 2023-09-01:
Gotcha, updated to omit those extra calls, since the problem wasn't ever very prominent for me and caused issues for you (and probably other people, if anyone's crazy enough to try and use this).

Good to hear it at least doesn't have any glaring issues after that fix, besides, yeah, the obvious under the hood.

The lack of configuration was more just a time constraint for how much time I want to spend on this. Not like it'd be hard to do, just relatively time consuming. I might update it in the future for that... (we'll see if I follow the philosophy of "if it ain't broke, break it")

Since I didn't want to deal with the tray and potential lack thereof, I've simply made it so that holding either "close" button for 5 seconds completely closes it.

Yeah... I don't care for the polling. For future reference, what would be the best way (or, a better way for this) to get events from spacenavd? I've seen the solution that SpaceMouse_Fusion360_Wine uses and that seems to just be using sockets, but still just polling in a while loop, unless I've overlooked something there (very possible)

I've also added the command line argument `--no-daemon`, per your suggestion.

jtsiomb, 2023-09-01:
Ideally you want to integrate the spacenavd socket handling into the same event loop which Qt uses to get events from the window system. What this does is let you block while waiting for events from any source, instead of asking again and again if there are any new events and wasting CPU time.

If Qt wasn't involved, it would be as simple as "call `select` to wait on all your sockets/file descriptors". See an example of that for a straight Xlib program in `fly.c` from the libspnav example "fly".

With Qt handling your event loop you need to ask Qt to monitor the spacenavd socket. Internally it boils down to the same thing, but you're not writing the event loop code, Qt has that. For an example of that see how "spnavcfg" does it in `main.cc` with a `QSocketNotifier`.

If you want a good book for UNIX systems programming, I suggest "Advanced Programming in the UNIX Environment" by the late Richard Stevens, which covers all these (not the Qt specifics, the system call level I/O multiplexing on UNIX) among other things.

ryryog25, 2023-09-01:
Gotcha, that makes sense to keep that all consolidated, if I get around to it, I'll definitely take a look at spnavcfg's implementation, and yeah, just looked at a preview for the book. I'm not one to typically turn to books for learning, but just glancing over it, seems very well-written and worthwhile (not to mention that if I actually want to get into learning more advanced concepts and practices I better get familiar with reading, hah)

Thanks again, I really appreciate your help!