FreeSpacenav/spacenavd #89: windows compatibility

ekaggrat, 2023-09-10:
is there a way to make this work in windows?

jtsiomb, 2023-09-10:
Windows is not currently supported, no.

closed 2023-09-10 by jtsiomb

BastelBaus, 2023-09-15:
I have the same question, maybe this works: https://spacenav.sourceforge.net/spnav-win32.html

jtsiomb, 2023-09-15:
I forgot about that. That was contributed by someone a long time ago as a potential windows counterpart to spacenavd, but it was pretty much immediately abandoned. I've no idea if it works and to what extent. You can probably still get it from the original svn repo mentioned in that page, but it's not part of the spacenav project, and not migrated with the move to github.

If anyone wants to contribute something similar in the future, and in order to become part of the spacenav project, it'll have to be integrated in the same codebase, not a separate thing in a different language, and any 3rd party library dependencies will have to be discussed and agreed upon first.

dewi-ny-je, 2024-10-03:
It's a pity because 3dconnexion dropped support for quite some perfectly working devices, but unfortunately this requires more funding than of course the people using 10+ years old devices would be interested in providing