FreeSpacenav/spacenavd #90: Space Mouse Pro: Received button 22 event on a 15 button device.

livelace, 2023-11-03:
spacenavd 1.2
libspnav 1.1

got this error with space mouse pro wireless:

```
Received button 22 event on a 15 button device.
This is a bug. Please report it:
https://github.com/FreeSpacenav/spnavcfg/issues
This warning will only be shown once.
```

meanwhile i have different devices and xmodmap -pp shows:

```
There are 20 pointer buttons defined.

    Physical        Button
     Button          Code
        1              1
        2              2
        3              3
        4              4
        5              5
        6              6
        7              7
        8              8
        9              9
       10             10
       11             11
       12             12
       13             13
       14             14
       15             15
       16             16
       17             17
       18             18
       19             19
       20             20

```

jtsiomb, 2023-11-07:
Could you post your `/var/log/spnav.log` (in pastebin if it's huge), and also could you follow the directions here: https://github.com/FreeSpacenav/spacenavd/wiki/Device-button-names#how-to-document-button-names and post a list of what number spacenavd thinks each button on your device is? In theory it should be like this for the spacemouse pro: https://github.com/FreeSpacenav/spacenavd/wiki/Device-button-names#spacemouse-pro, but obviously something is different, including the number of buttons (!)

livelace, 2023-11-08:
spacenavd doesn't have any buttons mappings in spnavrc

spacenavd.log:

```
Spacenav daemon 1.2
reading config file: /etc/spnavrc
found usb device [256f:c652]: "3Dconnexion 3Dconnexion Universal Receiver" (/dev/input/event4) 
adding device (id: 0).
device name: 3Dconnexion 3Dconnexion Universal Receiver
  Number of axes: 6 (6a 0r)
  Number of buttons: 15 (evdev offset: 256)
  Axis 0 value range: -350 - 350 (fuzz: 0)
  Axis 1 value range: -350 - 350 (fuzz: 0)
  Axis 2 value range: -350 - 350 (fuzz: 0)
  Axis 3 value range: -350 - 350 (fuzz: 0)
  Axis 4 value range: -350 - 350 (fuzz: 0)
  Axis 5 value range: -350 - 350 (fuzz: 0)
using device: 3Dconnexion SpaceMouse Wireless (guess) (/dev/input/event4)
  device flags: swap y-z invert y-z
trying to open X11 display ":1"
   XAUTHORITY=/root/.xauthbiJaov
Using XTEST to send key events
request SET_NAME - 706d6973 6520656c 706d6178 656c 0 0
client name: simple example
request DEV_NAME - 0 0 0 0 0 0
request DEV_PATH - 0 0 0 0 0 0
request DEV_NBUTTONS - 0 0 0 0 0 0
request DEV_NAXES - 0 0 0 0 0 0
adding dev event for device: /dev/input/event4
closing X11 connection to display ":1"
removing device: 3Dconnexion SpaceMouse Wireless (guess) (id: 0 path: /dev/input/event4)
removing pending device event of: /dev/input/event4
```

simple_af_unix.log:

```
spacenav AF_UNIX protocol version: 1
Device: 3Dconnexion SpaceMouse Wireless (guess)
Path: /dev/input/event4
Buttons: 15
Axes: 6

got button press event b(0)
got button release event b(0)
got button press event b(23)
got button release event b(23)
got button press event b(24)
got button release event b(24)
got button press event b(25)
got button release event b(25)
got button press event b(22)
got button release event b(22)
got button press event b(12)
got button release event b(12)
got button press event b(13)
got button release event b(13)
got button press event b(14)
got button release event b(14)
got button press event b(15)
got button release event b(15)
got motion event: t(0, 0, 0) r(0, 0, -4)
got motion event: t(-5, 0, 0) r(0, 0, -8)
got motion event: t(-3, 0, 0) r(0, 0, -5)
got motion event: t(-2, 0, 0) r(0, 0, -4)
got motion event: t(0, 0, 0) r(0, 0, 0)
got motion event: t(0, 0, 0) r(0, 0, 0)
got button press event b(8)
got button release event b(8)
got button press event b(2)
got button release event b(2)
got button press event b(26)
got button release event b(26)
got button press event b(5)
got button release event b(5)
got button press event b(4)
got button release event b(4)
got button press event b(1)
got button release event b(1)
```

simple_af_unix mapping:

```
menu - 0
alt - 23
shift - 24
ctrl - 25
esc - 22
1 - 12
2 - 13 
3 - 14
4 - 15
square/arrow - 8
square/t - 2dev-libs/libspnav
mouse/rotate - 26
square/f - 5
square/r - 4
fit - 1
```

jtsiomb, 2023-11-10:
Ok see what the problem is. For some reason, some of the newer 3Dconnexion devices are reporting discontinuous button numbers. I have added a low level remapping function in the driver, to map them to a contiguous range, which is applied to devices which are known to exhibit this issue.

The SpaceMouse Pro/Pro Wireless is known to need this remapping, and this remapping should be applied correctly when connected through a USB cable (btw could you verify that?). In wireless mode however, many wireless devices share the same device ID, and it's hard to distinguish between them, at least with the current evdev-based code. I have a heuristic to identify which is which, but apparently it fails to detect this as a SpaceMouse Pro Wireless, detecting it as a SpaceMouse Wireless instead, and fails to apply the remapping hack.

I'll see if I can devise a better heuristic to distinguish between spacemouse wireless and spacemouse pro wireless through the dongle. Could you post the output of `lsusb -d 256f:c652 -vvv` just in case it provides any hints?

For you using the device this bug shouldn't be a big obstacle. It should still work fine, and you could even remap the existing button numbers to whatever you need. The button remapping panel of `spnavcfg` will probably misbehave, but you should be able to remap them by manually editing the configuration file. I appreciate the help in trying to solve this however, and I'll get back at you if I need some more information about the device.

livelace, 2023-11-13:
>Could you post the output of lsusb -d 256f:c652 -vvv just in case it provides any hints?

```

Bus 001 Device 010: ID 256f:c652 3Dconnexion Universal Receiver
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x256f 3Dconnexion
  idProduct          0xc652 Universal Receiver
  bcdDevice            1.06
  iManufacturer           1 3Dconnexion
  iProduct                2 3Dconnexion Universal Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 3DX01.06_B0003
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     129
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
[... 90 lines cut ...]
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     129
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```

livelace, 2023-11-13:
>and this remapping should be applied correctly when connected through a USB cable (btw could you verify that?).

it looks like the usb connection brings correct buttons mappings. at least spnavcfg doesn't throw errors and buttons codes are not high. 

this is "cabled" version of lsusb -d 256f:c631 -vvv:

```
Bus 001 Device 016: ID 256f:c631 3Dconnexion SpaceMouse Pro Wireless (cabled)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x256f 3Dconnexion
  idProduct          0xc631 SpaceMouse Pro Wireless (cabled)
  bcdDevice            4.41
  iManufacturer           1 3Dconnexion
  iProduct                2 SpaceMouse Pro Wireless
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     365
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              20
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```

jtsiomb, 2024-07-12:
Can you try the current git version of spacenavd and let me know if the device is detected correctly and the buttons are mapped to a continuous range?

CtrlC-Root, 2024-07-13:
I have the same device and just tested the last release versus the last commit on the `master` branch. In the last release I ran into this issue and with the last commit on `master` it is now fixed. Tested with:

```bash
./configure
make
sudo ./spacenavd -d -l ./output.log
```

Here's the `output.log` file:

[output.log](https://github.com/user-attachments/files/16203905/output.log)

jtsiomb, 2024-07-14:
Excellent!

closed 2024-07-14 by jtsiomb