FreeSpacenav/spacenavd #91: kbmap not working under wayland

mseddon, 2024-01-14:
I am trying to support shift / alt / ctrl modifiers under wayland for Blender using wayland (without xwayland). Unfortunately this is not working for me- `xev` does not acknowledge any key events at all.

Am I correct in thinking this is unsupported at present, and that running xwayland is the right solution in the meantime?

jtsiomb, 2024-01-14:
Yes, the key mapping feature sends X events. It only works for X applications and only if spacenavd successfully connects to the X server, so any X authentication issues will prevent it from working too.

I'm marking this as a feature request, but I don't plan on implementing wayland keyboard mapping myself. I'm not at all interested in wayland, I consider it a step backwards, but if anyone else wishes to work on this, I'll gladly accept patches/pull requests about it.

adrianinsaval, 2024-11-18:
it might be possible to implement this using [libei](https://gitlab.freedesktop.org/libinput/libei) now

jtsiomb, 2024-11-19:
Unless I'm misreading the libei readme file, it sounds like the wayland compositor needs to support libei specifically for this to work. It feels like a kludge, and far from a universal solution for implementing "keystrokes on wayland".

Also no external libraries unless absolutely necessary.

adrianinsaval, 2024-11-19:
It's the most universal you can get in Wayland

blakeblair, 2025-02-06:
has anyone picked this up? I'm still holding out hope because blender doesn't seem to like the button numbers as inputs.

ftk, 2025-02-07:
See my example using uinput and libspnav
https://gist.github.com/ftk/0d70e527c05cc7c7d25bd57019a829a9

jtsiomb, 2025-02-07:
That's interesting. uinput seems like a viable way to generate fake keypresses independent of the window system. I'll look into it.