FreeSpacenav/spacenavd #94: Missing buttons on Magellan SpaceMouse Plus and XT Plus

Sebastien-GILBERT, 2024-02-26:
Dear,

The Magellan Serial SpaceMouse exist in 3 different versions: Classic, Plus and Plus XT .
The first one have only 9 buttons, but the two others have 11 (one button is added on each side of the cap).
Spacenavd only use the 9 buttons configuration, whatever which version is connected.

By default, the two cap's buttons (+ and -) are mapped to button 6 and 7, but if we set the extended key register, they become independent buttons.

It is explain in the programmer's guide:
- Link:		http://spacemice.org/pdf/Magellan_Programmers_Manual_2000.pdf
- Page:		6
- Command:	Compress Mode Set Command
- Bit:	extkey
= 1 State of the Magellan/SPACE MOUSE plus (+) button appears in bit B1 of nibble3 of the keyboard response (command "kQ\r"). State of the minus (-) button appears in bit B2 of nibble3.
= 0 State of the Magellan/SPACE MOUSE plus (+) button appears in bit B1 of nibble2 of the keyboard response (command "kQ\r"). State of the minus (-) button appears in bit B2 of nibble2.
				
The Windows official driver use this 11 buttons configuration for all Magellan SpaceMouse models.
- Link:		http://spacemice.org/pdf/Magellan_Manual_W_2000.pdf
- Page:		12
- Chapter:	Keyboard Mapping
For Magellan SPACE MOUSE Classic, the plus [+] and minus [-] buttons are present in the Keyboard Mapping window but have no effect.

				
To enabled them in Spacenavd, I have done the following modifications in dev_serial.c:
- Line 609:	Change the number of buttons from `9` to `11`
- Line 188:	Change init command from `write(fd, "m3\r", 3);` to `write(fd, "c3B\r", 4);`

I don't see any problem to use this setting by default on spacenavd.
Do you agree to commit this evolution ?

Best regards

jtsiomb, 2024-02-26:
What are the effects of doing this unconditionally, if there is a SpaceMouse classic connected? Are the "c" commands ignored by the classic? Otherwise we might have to detect which spacemouse is present, and change the initialization string accordingly. At the very least Your proposed change of the number of buttons on line 609 can't be hardcoded, and must depend on device detection.

The way the documentation describes these commands, it's not clear to me if the "c" command replaces the "m" command, or if both should be sent to the device. I think we should probably send both commands.

Finally your "c3B" string seems to enable the "Turbo Magellan" packet format, but the `mag_parsepkt` function expects the standard packet format. If I'm reading this correctly, either the command should be "c3A", or the packet handling function should be modified to handle either format.

Sebastien-GILBERT, 2024-03-01:
As the programmer's guide is common to the three models (see its cover), to me there is no side effect to use the "c" command on classic mouse.
The windows driver hardcoded the activation of the 11 buttons on the three versions.

To my understanding, the "m" command [operating mode] was for the first version of the protocol (and kept for compatibility) and the "c" command [extended operating mode] take all the "m" command parameters and add new ones. And so, the "c" command is enough for a Magellan mouse.

The "c3B" string don't activate the "turbo mode", only the extended keys
What we want for the [Nibble2]:
- [B3] not define --> 0
- [B2] Quick tip: off --> 0
- [B1] extkey: activated --> 1
- [B0] compress: off --> 0

So the Nibble value is "0010". According to the conversion table page 5, it correspond to the character "B".
The string "c3A" correspond to the nibble "0001" and in this case activate the compress mode without the extended keys.

jtsiomb, 2024-03-01:
You're right of course, I forgot the encoding table and treated it as binary in my mind.

About the commands, you're probably right about all devices described in that document supporting those commands, but I can't remember off-hand the lineage of the magellan devices and what were the previous models if any. Presumably *some* device originally didn't have the extended init command, and I'd rather be compatible by default if possible. In any case since we need to detect the device in order to fill the number of buttons field in the device structure, it doesn't make much sense to send the extended command to activate the extra buttons if the device doesn't have them.

So I'll try to do something about detecting the spacemouse plus, and I'll get back at you for further testing. Which of those devices do you have and are able to test with?

Sebastien-GILBERT, 2024-03-01:
I can't see how you can discover the model, my SpaceMouse is a Plus and report "v  MAGELLAN  Version 5.79  by LOGITECH INC. 10/10/97" in the driver.
I can do test only on a spacemouse Plus and I an on a Debian 12 computer.

My spacenavd.log
```
Spacenav daemon v1.2-10-ga2d121a
reading config file: /etc/spnavrc
adding device (id: 0).
Magellan SpaceMouse detected:
v  MAGELLAN  Version 5.79  by LOGITECH INC. 10/10/97

11 buttons
using device: /dev/ttyS0
trying to open X11 display ":0.0"
   XAUTHORITY=/root/.Xauthority
failed to open X11 display ":0.0"
waiting for the X socket file to appear
adding dev event for device: /dev/ttyS0
```

jtsiomb, 2024-03-20:
I just added the extended buttons enable command to the magellan init, and changed the button count from 9 to 11.
I did not remove the `m` command, just in case previous firmware does not recognize the `c` command. I'm assuming that if it doesn't, it will ignore the c command and continue with `m3` in effect.

closed 2024-03-20 by jtsiomb