FreeSpacenav/spacenavd #95: can not find it

Nimingez, 2024-02-27:
[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python /home/nmz/robosuite/robosuite/scripts/setup_macros.py (macros.py:55)
DataCollectionWrapper: making new directory at /tmp/1709036351_847035
Opening SpaceMouse device
Traceback (most recent call last):
  File "collect_human_demonstrations.py", line 241, in <module>
    device = SpaceMouse(pos_sensitivity=args.pos_sensitivity, rot_sensitivity=args.rot_sensitivity)
  File "/home/nmz/robosuite/robosuite/devices/spacemouse.py", line 126, in __init__
    self.device.open(self.vendor_id, self.product_id)  # SpaceMouse
  File "hid.pyx", line 142, in hid.device.open
OSError: open failed

can not find spacemouse
systemctl status spacenavd
● spacenavd.service - 3Dconnexion Input Devices Userspace Driver
     Loaded: loaded (/lib/systemd/system/spacenavd.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-02-27 20:17:59 CST; 4min 29s ago
   Main PID: 988 (spacenavd)
      Tasks: 1 (limit: 76703)
     Memory: 964.0K
     CGroup: /system.slice/spacenavd.service
             └─988 /usr/bin/spacenavd -d -v

2月 27 20:18:15 nmz-desktop spacenavd[988]: already using device: 3Dconnexion 3Dconnexion Universal Receiver (/dev/input/event4)
2月 27 20:18:15 nmz-desktop spacenavd[988]: failed to find any supported devices

CtrlC-Root, 2024-02-27:
I'm guessing you're running this code based on the stack trace above:
https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/devices/spacemouse.py#L126

It looks like it's looking for these hard-coded vendor and product IDs:
https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/macros.py#L37

Those don't match the USB receiver:

```
$ lsusb
Bus 003 Device 007: ID 256f:c652 3Dconnexion Universal Receiver
$ python
Python 3.12.2 (main, Feb 15 2024, 06:47:30) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 0x256f
9583
>>> 0xc652
50770
```

It seems you may need to change the vendor and product IDs the code is using as per the instructions at:
https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/devices/spacemouse.py#L11.

Did you try that? Also does spnavcfg work for you?

Nimingez, 2024-02-27:
yes I am running robosuite.
I have changed the ID
# Spacemouse settings. Used by SpaceMouse class in robosuite/devices/spacemouse.py
SPACEMOUSE_VENDOR_ID = 9583
SPACEMOUSE_PRODUCT_ID = 50770

Nimingez, 2024-02-27:
nmz@nmz-desktop:~$ spnavcfg
spnavcfg: symbol lookup error: spnavcfg: undefined symbol: spnav_cfg_set_repeat

jtsiomb, 2024-02-27:
This bug report doesn't seem to have anything to do with spacenavd, but rather about using the spacemouse directly through hidapi. Feel free to re-open the bug report if I misunderstood, and try to explain clearly how it is relevant to spacenavd.

About your spnavcfg build issue: you seem to be trying to build post-1.1 spnavcfg code from git, which won't build with libspnav 1.1. You either need to build spnavcfg 1.1, or use the current libspnav from git as well.

closed 2024-02-27 by jtsiomb