FreeSpacenav/spacenavd #97: spacenav with ch341 usb-serial support?

nimishgautam, 2024-03-11:
Hey, I'm using a CH341 serial-to-usb converter device to try and run an old serial spacemouse (using the latest driver for the adapter's chipset). 

I've tried setting the spacenavd serial parameter to /dev/ttyCH341USB0 (the virtual device that the adapter creates), but when it loads up, the spacemouse makes a loud beeping noise for a few seconds and then that's it. I didn't even know the thing had any parts that would make noise, so I'm at a bit of a loss on how to debug this.

Neither simple_af_unix or simple_x11 produce any output, and spnavcfg just says '6-dof-device' but seems to be a generic view.

Here's the output of the spnavd.log

``` 
Spacenav daemon v1.2-11-g1d6fde6
reading config file: /etc/spnavrc
adding device (id: 0).
removing device:  (id: 0 path: /dev/ttyCH341USB0)
Using XTEST to send key events
request SET_NAME - 616e7073 67666376 0 0 0 0
client name: spnavcfg
request SET_EVMASK - ffff 0 0 0 0 0
request DEV_NAME - 0 0 0 0 0 0
request DEV_PATH - 0 0 0 0 0 0
request DEV_NBUTTONS - 0 0 0 0 0 0
request DEV_NAXES - 0 0 0 0 0 0
request DEV_TYPE - 0 0 0 0 0 0
request GCFG_SENS - 0 0 0 0 0 0
request GCFG_SENS_AXIS - 0 0 0 0 0 0
request GCFG_INVERT - 0 0 0 0 0 0
request GCFG_SWAPYZ - 0 0 0 0 0 0
request GCFG_LED - 0 0 0 0 0 0
request GCFG_GRAB - 0 0 0 0 0 0
request GCFG_AXISMAP - 0 0 0 0 0 0
request GCFG_DEADZONE - 0 0 0 0 0 0
request GCFG_AXISMAP - 1 0 0 0 0 0
request GCFG_DEADZONE - 1 0 0 0 0 0
request GCFG_AXISMAP - 2 0 0 0 0 0
request GCFG_DEADZONE - 2 0 0 0 0 0
request GCFG_AXISMAP - 3 0 0 0 0 0
request GCFG_DEADZONE - 3 0 0 0 0 0
request GCFG_AXISMAP - 4 0 0 0 0 0
request GCFG_DEADZONE - 4 0 0 0 0 0
request GCFG_AXISMAP - 5 0 0 0 0 0
request GCFG_DEADZONE - 5 0 0 0 0 0
request GCFG_BNMAP - 0 0 0 0 0 0
request GCFG_BNACTION - 0 0 0 0 0 0
request GCFG_KBMAP - 1 0 0 0 0 0
request GCFG_SERDEV - 0 0 0 0 0 0
request GCFG_SERDEV - 0 0 0 0 0 0
request SCFG_SERDEV - 7665642f 7974742f 34334843 42535531 30 0
request GCFG_SENS - 0 0 0 0 0 0
request GCFG_SENS_AXIS - 0 0 0 0 0 0
request GCFG_INVERT - 0 0 0 0 0 0
request GCFG_SWAPYZ - 0 0 0 0 0 0
request GCFG_LED - 0 0 0 0 0 0
request GCFG_GRAB - 0 0 0 0 0 0
request GCFG_AXISMAP - 0 0 0 0 0 0
request GCFG_DEADZONE - 0 0 0 0 0 0
request GCFG_AXISMAP - 1 0 0 0 0 0
request GCFG_DEADZONE - 1 0 0 0 0 0
request GCFG_AXISMAP - 2 0 0 0 0 0
request GCFG_DEADZONE - 2 0 0 0 0 0
request GCFG_AXISMAP - 3 0 0 0 0 0
request GCFG_DEADZONE - 3 0 0 0 0 0
request GCFG_AXISMAP - 4 0 0 0 0 0
request GCFG_DEADZONE - 4 0 0 0 0 0
request GCFG_AXISMAP - 5 0 0 0 0 0
request GCFG_DEADZONE - 5 0 0 0 0 0
request GCFG_BNMAP - 0 0 0 0 0 0
request GCFG_BNACTION - 0 0 0 0 0 0
request GCFG_KBMAP - 0 0 0 0 0 0
request GCFG_BNMAP - 1 0 0 0 0 0
request GCFG_BNACTION - 1 0 0 0 0 0
request GCFG_KBMAP - 1 0 0 0 0 0
request GCFG_SERDEV - 0 0 0 0 0 0
request GCFG_SERDEV - 0 0 0 0 0 0
```

jtsiomb, 2024-03-11:
I have tested serial spaceballs and magellan spacemice with USB-serial adapters and it generally works. I don't think I have a ch341 adapter though. All of mine are based on the pl2303 coincidentally.

The "removing device" line implies that it tried to open it but failed to communicate correctly with any of the two possible serial protocols, and gave up.

This might very well be an issue with the ch341 adapter or its driver, if it failed to set up the tty correctly. Especially since you said you're trying to use a spacemouse. The magellan spacemouse serial protocol uses two stop bits, something which is not very common, and could easily not be supported by the adapter at all. If you want to try an experiment, open `src/dev_serial.c` find the `stty_mag` function, and remove `CSTOPB` from the OR-ed terminal control flags, see if it makes a difference.

Btw which model of spacemouse do you have exactly?

nimishgautam, 2024-03-11:
Thanks for the suggestion :)

I tried removing the flag (and recompiling, reinstalling etc) but got the same result as before. It makes some loud beeping noises and that's it. This is the log output, nothing else.

```
Spacenav daemon v1.2-11-g1d6fde6
reading config file: /etc/spnavrc
adding device (id: 0).
removing device:  (id: 0 path: /dev/ttyCH341USB0)
Using XTEST to send key events
```

I'm not sure which version I have, it just says "Magellan/SpaceMouse" underneath, has buttons 1-8 and a * on top, 2 side buttons, and a serial port.  It's the same one from this hackaday except a different color: https://github.com/jfedor2/magellan-spacemouse

I was hoping it'd be possible to run it without all the layers of conversion.

jtsiomb, 2024-03-11:
I was about to say that I have the same spacemouse plus here, and it worked with my USB-serial before, but evidently there's either some regression, or I'm misremembering and I only ever used it with a real serial port, because it doesn't work reliably for me either right now. I'll do some more testing and get back at you.

jtsiomb, 2024-03-12:
Ok I did some more testing. With a different USB-serial adaptor, it worked perfectly, while with the one I tested earlier today I was always getting garbled communications. Both adaptors with the same pl2303 chip btw.

I'm afraid there's nothing I can do in spacenavd to make a bad adaptor correctly transmit and receive the somewhat uncommon 9600 8N2 (2 stop bits) required by the magellan devices. Spaceballs are easier to get to work with shitty USB-serial adaptors, because they use the defacto standard 8N1 framing. I still like magellan spacemice more though to be honest, the ball feels more clumsy.

I suggest either switching to your computer's built-in "real" serial port if you have one (most desktop PCs do, if only as a pin header on the motherboard where you can plug a serial port bracket), or try your luck with a different USB-serial adaptor. For the best chance of finding one that works, I'd look for an FTDI FT232-based one.

As this doesn't seem to be a bug in spacenavd, I'm closing the bug report. But feel free to re-open it if you find any evidence to the contrary. And by all means also let us know if you find another USB-serial adaptor which works.

closed 2024-03-12 by jtsiomb