FreeSpacenav/spacenavd #98: spacenavd needs authorization?

mcdavich, 2024-04-25:
the spacenavd is running but requies authorization?  How do I authorize it?

sudo systemctl status spacenavd

spacenavd.service - 3Dconnexion Input Devices Userspace Driver
     Loaded: loaded (/lib/systemd/system/spacenavd.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-04-25 14:52:34 EDT; 11s ago
       Docs: man:spacenavd(8)
   Main PID: 16966 (spacenavd)
      Tasks: 1 (limit: 76839)
     Memory: 368.0K
        CPU: 4ms
     CGroup: /system.slice/spacenavd.service
             └─16966 /usr/bin/spacenavd -d -v

Apr 25 14:52:34 debian systemd[1]: Started spacenavd.service - 3Dconnexion Input Devices Userspace Driver.
Apr 25 14:52:34 debian spacenavd[16966]: Authorization required, but no authorization protocol specified

jtsiomb, 2024-04-25:
authorization? I've no idea where this error message is coming from. Is systemd attempting to run spacenavd as an unpriviledged user? spacenavd needs to run as root.

mcdavich, 2024-04-27:
Turns out the application I was attempting to control threw the error.  Apologies for the confusion.

closed 2024-04-27 by mcdavich