FreeSpacenav/spacenavd #101: Scroll up/down ?

wawanbreton, 2024-05-14:
Hi,
I just got a SpaceMouse Enterprise at work, and trying to see if I can use it actively for software development.
Is there an easy way to map the up-down move to a mouse wheel ? That would allow me to very easily scroll through large source files.
By the way, I'm very happy to see that this project is still active :)

jtsiomb, 2024-05-14:
There is no way currently to map analog axes to generate fake X11 events. It could be done, but I'm not entirely sure it would be useful. If you mapped the vertical axis to generate mouse wheel events, you couldn't use it for its intended purpose which is navigating in 3D viewports.

I think the best way to do this, is to hack your text editor to add 6dof input support with libspnav, and use the translation events to move the scroll area proportionally to the magnitude of the Y translation events.

wawanbreton, 2024-05-14:
Hmm that's an interesting idea indeed, I'm using Qt Creator which has a plugin system, so that should be doable **quite** easily :smiley: I'll give it a thought, thanks !

ftk, 2024-05-22:
I've made a simple uinput driver that sends fake mouse/keyboard events using spacenav motion: https://gist.github.com/ftk/0d70e527c05cc7c7d25bd57019a829a9

wawanbreton, 2024-05-22:
> I've made a simple uinput driver that sends fake mouse/keyboard events using spacenav motion: https://gist.github.com/ftk/0d70e527c05cc7c7d25bd57019a829a9

Interesting, I should check that.

In the meantime, I have made a Qt Creator plugin, which was indeed very easy, and works very well. But not really useful in the end :laughing: Anyway, I will still publish it someday.

closed 2024-05-22 by jtsiomb