FreeSpacenav/spacenavd #104: Source code uses "struct timeval" which is 2038-unsafe, causes build failures on 32-bit systems

hosiet, 2024-06-01:
In https://github.com/FreeSpacenav/spacenavd/blob/39856625a6de1e8c4b57c5938e1bf29d13cf1a9f/src/event.h#L89-L94 :

The spacenavd source code uses `struct timeval`, which is not 2038-safe. This field is retrieved from linux evdev event, as below:

https://github.com/FreeSpacenav/spacenavd/blob/39856625a6de1e8c4b57c5938e1bf29d13cf1a9f/src/dev_usb_linux.c#L271-L293

However, `struct input_event` as from Linux headers is removing the `.time` field on 32-bit systems due to Y2038 problem. You may find the initial Linux kernel patch at https://lkml.org/lkml/2018/1/6/324 , and a similar report from https://bugreports.qt.io/browse/QTBUG-84012 .

This problem is causing build failures in Debian. You may find details in https://bugs.debian.org/1069474 .

* * *

The solution should be no longer using `.time` as `struct timeval`, but to use `.input_event_sec` and `.input_event_usec` fields instead.

jtsiomb, 2024-06-01:
Thanks for pointing this out. I'll look into it asap.

jtsiomb, 2024-06-02:
Turns out that time value was not really used very much, and the only place which used it shouldn't be using it (see issue #84). So I just dropped it altogether.

closed 2024-06-02 by jtsiomb