FreeSpacenav/spacenavd #105: Build failure with musl libc

CtrlC-Root, 2024-06-16:
I'm attempting to update the spacenavd package in Void Linux to the latest v1.3 release and I encountered a build error when using musl libc. The [void linux package pull request](https://github.com/void-linux/void-packages/pull/50849) has all the automated builds and you can look at the [x86_64-musl](https://github.com/void-linux/void-packages/actions/runs/9536782450/job/26284104445?pr=50849) one as a representative example. I've been able to reproduce this failure on my local machine as well. It seems to be an issue with including `stdint.h` correctly. I'm going to investigate this and see if I can figure out the fix but it's not immediately obvious to me so I thought I'd report it here in case it is immediately obvious to someone else.

CtrlC-Root, 2024-06-16:
Actually I may have figured it out. Here's the `config.h` file that is generated by the configure script:

```
#ifndef CONFIG_H_
#define CONFIG_H_

#define USE_X11

#define USE_NETLINK

#define VERSION "v1.3"

#define HAVE_ALLOCA_H
#define HAVE_MALLOC_H
#define HAVE_STDINT_H
#define HAVE_INTTYPES_H

#define CFGDIR "/etc"

#endif	/* CONFIG_H_ */
```

It seems to be defining `HAVE_STDINT_H` whereas the source files (ex: `src/proto.h` line 6) are checking for `HAVE_STDINT_H_` instead. Unless I'm missing something basic that seems like the issue.

CtrlC-Root, 2024-06-16:
This patch fixes the build: https://github.com/void-linux/void-packages/pull/50849/commits/267e893f47803780cc077e71037fe184d356a184#diff-1a398d29bd96ad0905106bf6a24e0b62386928d9641dab2e3e4b3b282f429ca2

jtsiomb, 2024-06-16:
Indeed, thanks for the report. Fixed.

closed 2024-06-16 by jtsiomb

jtsiomb, 2024-06-16:
@CtrlC-Root I just took a look at your CI. If this is indicative of how you compile binary packages for the void linux repo, you're building spacenavd without XTEST and XInput2 support as indicated by the configure script not finding the appropriate header files. XInput2 might not be necessary any more (it's used by a bit of code to make sure to grab the device away from the X server if it happens to want to use it as a regular mouse, which it used to do in the past). The XTEST extension however is needed for generating keyboard events if the user binds keystrokes to device buttons. The altenrative fallback to XSendEvent is not reliable.

CtrlC-Root, 2024-06-16:
> @CtrlC-Root I just took a look at your CI. If this is indicative of how you compile binary packages for the void linux repo, you're building spacenavd without XTEST and XInput2 support as indicated [...]

Ah, thanks for letting me know. I didn't notice as I don't use this feature and didn't think to test for it. I'll look into fixing that next.

CtrlC-Root, 2024-06-22:
Hmm, as far as I can tell, it's finding the header files. A few I spot checked:

https://github.com/void-linux/void-packages/actions/runs/9536782450/job/26284104121?pr=50849#step:7:77
https://github.com/void-linux/void-packages/actions/runs/9536782450/job/26284104445?pr=50849#step:7:62

Why do you believe it's building without XTest support? Did I miss something in the build output?

CtrlC-Root, 2024-06-22:
Oh I see, because `HAVE_XTEST_H` is not in the `config.h` file, right?

CtrlC-Root, 2024-06-22:
It seems to be linking against the library though:

https://github.com/void-linux/void-packages/actions/runs/9536782450/job/26284104121?pr=50849#step:7:167

On my machine:

```
alex@hammond ~/w/c/void-packages (spacenavd-xtest-support)> ldd (which spacenavd)
        linux-vdso.so.1 (0x00007fff8f8a5000)
        libXi.so.6 => /usr/lib/libXi.so.6 (0x00007fc6644a5000)
        libXtst.so.6 => /usr/lib/libXtst.so.6 (0x00007fc66449c000)
        libX11.so.6 => /usr/lib/libX11.so.6 (0x00007fc664359000)
        libXext.so.6 => /usr/lib/libXext.so.6 (0x00007fc664344000)
        libc.so.6 => /usr/lib/libc.so.6 (0x00007fc66415a000)
        libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007fc664130000)
        /lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007fc6644ec000)
        libXau.so.6 => /usr/lib/libXau.so.6 (0x00007fc664129000)
        libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007fc664121000)
```

I'm not sure why it's not set in the `config.h` file above.

jtsiomb, 2024-06-23:
The one I saw the other day was saying that it didn't find either of those libraries. I don't know which run that was though. This one you posted now certainly finds them so that's perfectly fine.