FreeSpacenav/spacenavd #110: KP_Space as kbmap0 gives unexpected X error: BadValue

mw75, 2024-07-12:
I'm running the latest version of spacenavd with X11 and libXTEST with a spacepilot. 
I would like to map KP_Space to to button 1 - index 0. Unfortunately this throws an error and the keystroke is not emitted. I already tried to fallback to XSendEvent, but that also did not work.

Running in foreground i get:
Spacenav daemon v1.3-1-gc6f8a6b
reading config file: /etc/spnavrc
Device detection, parsing /proc/bus/input/devices
found usb device [46d:c625]: "3Dconnexion SpacePilot" (/dev/input/event8) 
adding device (id: 0).
device name: 3Dconnexion SpacePilot
  Number of axes: 6 (0a 6r)
  Number of buttons: 21 (evdev offset: 256)
using device: 3Dconnexion SpacePilot (/dev/input/event8)
  device flags: swap y-z invert y-z
trying to open X11 display ":0"
   XAUTHORITY=/home/mw75/.Xauthority
Using XTEST to send key events
unhandled event: 17
unhandled event: 17
mapping ``KP_Space'' to keysym 65408
Caught unexpected X error: BadValue (integer parameter out of range for operation) [op: 132,2, res: 0]
Caught unexpected X error: BadValue (integer parameter out of range for operation) [op: 132,2, res: 0]

KP_Delete on kbmap1 and everything else works fine!

PS: is it possible to control the LCD somehow?

jtsiomb, 2024-07-12:
KP_Space (keysym 65408 or ff80h) is not space, it's "keypad space" which probably doesn't exist on your keyboard layout, and the X server treats it as invalid. Try mapping "space" instead, which is keysym 32 (20h).

jtsiomb, 2024-07-14:
I'll assume that was the problem, and close the issue. Feel free to reopen if turns out it's not, and it is a bug in spacenavd after all.

closed 2024-07-14 by jtsiomb